slic3r/Slic3r #3742: What is the release plan?

freiser77, 2017-03-07:
As the prusa3d-fork is releasing fairly often, I would appreciate a close merging of prusa3d's features (when possible) in the main repo.

On the other hand, the last release of the main repo was on 17 Jun 2015.
Are any further releases planned or is the prusa3d-fork the main one in future and this one abandoned?

alexrj, 2017-03-07:
Unfortunately no pull requests are being submitted from that fork, so we try to pick what we can.

On the other hand there has been a LOT of work on the main project since the last release, and that fork is of course based on this work.

We are working for releasing a stable 1.3.0 version, but in the mean time you can find automated builds from this repository:
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/view

closed 2017-03-07 by alexrj

freiser77, 2017-03-07:
@alexrj: But what is in the way of releasing 1.3.0? How can we release more often?
Otherwise more and more people are using prusa3d's fork (me included) and this one is merely dead.

lordofhyphens, 2017-03-07:
Look at the milestone tracker. When that is cleared there will be an
official release.

Until then, we have fresh builds of Slic3r for OSX and Windows for people
to use in the meantime, automatically generated as new stuff gets added.

We port the prusa3d work when it looks useful to us. If they submitted pull
requests it would probably happen more often.

alexrj, 2017-03-07:
@freiser77, use the automated builds. They contain all the goodness of this 1.5 years of development. :)