slic3r/Slic3r #3744: Questions about windows-build.txt

gshasdy, 2017-03-08:
### Version
master branch

### Operating system type + version
Windows 7 64bit
Strawberry Perl

### Behavior
There's a windows-build.txt under the src folder with some instructions about building slic3r with mingw32-make, I don't think this part is mentioned or documented in the guide/wiki, so I'd like to know if this is updated and useful? 

The reason I'm asking is I'd like to debug the C++ code step by step in order to understand the flow. I'm pretty new to Perl, so I haven't found a solution to debug the code compiled with XS, any suggestions? 

Thank you!

bubnikv, 2017-03-08:
The simplest would be to pull my Perl binary distros compiled with Visual Studio 2013, install Visual Studio 2013 + Boost binaries, compile Slic3r by running perl Build.PL, run the process, attach Visual Studio 2013 debugger to the process and so on.

The links to my distros are at the top of the following page. You may volunteer to improve the page once you collect experience with it :-)
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-on-MS-Windows

bubnikv, 2017-03-08:
> https://github.com/prusa3d/Slic3r/wiki/How-to-compile-on-MS-Windows

The page is far from complete, currently it is only a random set of links.

You are supposed to extract my perl pre-build distro to c:\wperl64 or c:\wperl32, for the debugging build to c:\wperl64d
then to extract the binary boost to c:\local
then to extract the binary intel tbb-2017_U5.7z to c:\local
then to start the command line window by running c:\wperl64\cmdline.lnk
which will open a command window with everything pre-configured. Just pull and compile Slic3r, the gui libraries are already there.

lordofhyphens, 2017-03-08:
That file is, as best as I can see,  related to xsgui and the C++ CLI @alexrj was working on in the src dir. You can *probably* ignore it.

gshasdy, 2017-03-08:
Thank you so much for the instruction. I've got some errors when compiling Slic3r with the pre-compiled packages. I'll post my issue under the prusa3d branch. 

My goal is to understand the flow and add a custom feature which generates a SVG file for a given Z-index instead of all the layers.  Ideally I'd like to modify this portion of the C++ code and merge into my own C++ project. Would anyone please help me to point out which portion of the C++ code would be related to this and which file would be the entry point?

Thanks!

lordofhyphens, 2017-03-08:
If you use Strawberry Perl 5.18.4.1 64-bit as your general environment on Windows, you can dump the local-lib dir extracted from https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-perl-dependencies-5.18.0-win-x64.7z. This will take care of the Perl dependencies and is the same archive I use on the build server. That will get you at least somewhere. 

@bubnikv's custom perl/MSVC++ setup *should* work with upstream, but @alexrj and I don't do our primary development work on Windows (IIRC @alexrj is on OSX and I'm on Debian Linux).

https://github.com/alexrj/Slic3r/tree/master/xs/src/libslic3r take a look at SVG.cpp and SVG.hpp and work your way out from there.

This is a copy from the environment setup on the windows build server (it's in Powershell format). Obviously you will need to change some paths around, but it works every day, every commit.

There's some known issues with having multiple compiler setups in your path (MSVC, GCC, etc).  My solution has to remove the extra ones.

Be *very* careful with this script, make sure it doesn't delete anything you want to keep.
````
mkdir C:\projects\slic3r\FreeGLUT
if (!(Test-Path "C:\users\appveyor\freeglut.7z")) 
{
wget "https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=freeglut-mingw-3.0.0.win64.7z" -o C:\users\appveyor\freeglut.7z
}
cmd /c "7z x C:\Users\appveyor\freeglut.7z -oC:\projects\slic3r\FreeGLUT"

if (!(Test-Path "C:\users\appveyor\strawberry.msi")) {
wget "http://strawberryperl.com/download/5.18.4.1/strawberry-perl-5.18.4.1-64bit.msi" -o "C:\users\appveyor\strawberry.msi" | Write-Output
cmd /c "msiexec.exe /quiet /qn /norestart /i C:\Users\appveyor\strawberry.msi"
}
$PERLDIR = 'C:\Strawberry'
$env:Path = "C:\Strawberry\c\bin;C:\Strawberry\perl\bin;C:\Strawberry\perl\vendor\bin;" + $env:Path
if(Test-Path -Path 'C:\Strawberry' ) {
copy C:\Strawberry\c\bin\gcc.exe C:\Strawberry\c\bin\cc.exe
cmd /c mklink /D C:\Perl C:\Strawberry\perl
mkdir C:\dev
if (!(Test-Path "C:\users\appveyor\boost.1.59.0.7z")) {
cmd /c mklink /D C:\dev\boost_1_59_0 C:\Libraries\boost_1_59_0
cd C:\dev\boost_1_59_0
./bootstrap.bat mingw
./b2 --clean-all
./b2 toolset=gcc link=static runtime-link=static variant=release
} else {
cmd /c "7z x C:\Users\appveyor\boost.1.59.0.7z -oC:\dev"
}
mkdir C:\dev\CitrusPerl
cmd /C mklink /D C:\dev\CitrusPerl\mingw32 C:\Strawberry\c
cd C:\projects\slic3r
cpanm ExtUtils::Typemaps::Basic
cpanm ExtUtils::Typemaps::Default
cpanm local::lib
rm -r 'C:\Program Files\Git\usr\bin' -Force
}if (!(Test-Path "C:\users\appveyor\local-lib.7z")) {
wget "https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-perl-dependencies-5.18.0-win-x64.7z" -o "C:\users\appveyor\local-lib.7z" | Write-Output
}
if (Test-Path "C:\users\appveyor\local-lib.7z") {
cmd /c "7z x C:\Users\appveyor\local-lib.7z -oC:\projects\slic3r" -y | Write-Output
}

if (Test-Path "C:\Users\appveyor\boost.1.59.0.7z") {
$env:Path = "C:\Strawberry\c\bin;C:\Strawberry\perl\bin;" +     $env:Path
  perl Build.pl | Write-Output
}
if (!(Test-Path "C:\Users\appveyor\boost.1.59.0.7z")) {
cmd /c "7z a C:\users\appveyor\boost.1.59.0.7z C:\dev\boost_1_59_0 -xr!*.o -xr!*lib*-msvc-*"
}
````

bubnikv, 2017-03-08:
By the way, the Visual Studio 2013 builds are around 10% faster. Not much, but it counts.

gshasdy, 2017-03-08:
Thanks for the comments.

closed 2017-03-08 by gshasdy