slic3r/Slic3r #3746: Skirt taller than 1 is not getting proper adhesion

Tinchus2009, 2017-03-09:
### Version
all

### Operating system type + version
win10 but guess all 

### Behavior

When printing ABS, anything against drafts will help to avoid warping. The ability to have the skirt a value as tall as the object is GREAT for tall ABS objects, in my experience I could say it is the difference between getting a a good end and getting a warped piece.
The probem right now is that when you set the skirt to be as tall as the object, a protective wall is built arounf the object, but this wall will suffer of warping the same as the object. Since this skirt has no brim, or any other way of getting a good adhesion to the bed, for sure it will warp once the printing has reached 10 cms tall, almost for sure it will.

The solution would be to have the ability to include a brim to this skirt or when you set the brim teh external line of the brim to become this skirt and so skirt and brin will be one, giving to the skirt the necesary adhesion
_Is this a new feature request?_  YES

I tried setting a skirt distance of CERO, so the skirt is atached to the object  but this option is not working (is like the cero distance is just ignored). Anyway, this would not be a solution because the object would have an extra skin...

lordofhyphens, 2017-03-09:
Someone had a hack that permitted printing the skirt on top of the brim.

Tinchus2009, 2017-03-10:
Ok, I wasnt able to find that hack, but anyway, wouldnt be better to have this option available. Definitly is an issue that will happen most of the time when useing this kind of skirts. Haveing something like an option to set the skirt to be at the end of the brim (last line of the brim would atack the skirt to it and so give the adherence that needs to have) or have the option to position the skirt in the middle of the brim for example. Right now you you position the pointer on top of the skirt separation field you get a message saying that setting the separation to 0 would give you a brim. This behaviour is not happening. If you set the skirt separatio to 0, this value is ignored and a separation of 1 mm is introduced, may be makeing a cero value means that the skirt is atached to the end of the brim would work?

DrLex0, 2017-03-12:
What I usually do, is increase the minimum extrusion length until the skirt has a decent number of loops on its first layer, which effectively gives it a brim. Contrary to what the preview shows, only the first layer will have these extra loops, the other layers only have one loop (or whatever is set as minimum). Not sure if this is a bug in itself and the preview shows the intended behavior, but it does come in handy.

Tinchus2009, 2017-03-13:
I dont know if this is the hack @lordofhyphens was talking about but it works and it does it realy well!!!!
And no, it is not a bug, acording to the instructions, the correct behaviour if you set a minimun length of extruded material for the skirt, this value will be taken into consideration only for the bottom layer, that is the first layer I think. 
The bug actually it is in the 3d preview, the extra lines should not appear in the next layers.
@lordofhyphens , @alexrj , should I create a new report for this? The bug is:

if you set a minimun lenght to be extruded for the skirt, and this minimun lenght makes the skirt to have a couple of lines around the object, in the real print this extra lines only happens in the first layer (wich is the behaviour I was looking for, to have a brim for the skirt). But in the 3D preview, these extra lines appear in the hole skirt (assuming the skirt is taller than 1 layer).

Thanks @DrLex0 for the tip!!!!!

lordofhyphens, 2017-03-24:
@Tinchus2009 Well, if it's useful it's a feature, if not it's a bug :D

Tinchus2009, 2017-03-31:
Super usefull feature. What it is a bug is the 3d simulation, in the simulation the skirt is shown to be as tall as the object