slic3r/Slic3r #3753: How to arrange multi color STL models

Massa0815, 2017-03-11:
### Version
1.2.9

Dual Extrusion

### Operating system type + version
Windows 7

### Behavior
In older Versions there was a Menu item for combining STL FIles to an AMF File. In the actual Version 1.2.9 there is only the posibility to export the actual plate to .amf . But how can i combine two stl without coordinate offset. If i import two stls from a dual colored model, they are not at the correct position. How can i shift them exactly together, Especially in Z direction?

alexrj, 2017-03-11:
* Add the first part to the plater
* Double click it
* Click on "Load part..." and choose the second part

Voila, you just made a multi-part object. Make sure the two STL files share the same coordinate origin, and they will be automatically aligned.

closed 2017-03-11 by alexrj

Massa0815, 2017-03-11:
Very nice. It works... You saved my Day ;) 
I dint find it in your documentation

stuartpb, 2018-02-03:
I just tried this, and the parts are still getting added misaligned (Slic3r keeps introducing transforms I can't get rid of).

lordofhyphens, 2018-02-03:
@Stuartpb are you sure both models have the same reference 0 point?

Turn off auto arrange x/y and add both to the plater manually. If they are misaligned then it's the files.