slic3r/Slic3r #3754: Cannot build slic3r on Windows 7, stops at installing ./xs

reeseallen, 2017-03-11:
### Version
1.2.9-706-g5d10d4c0

### Operating system type + version
Windows 7

### Behavior
Cannot build slic3r from source

Please forgive my ignorance/naivete. I have little programming knowledge and this is my first time trying to download source code from git and compile it with Perl. Several hours of googling has gotten me this far, but I'm getting no further on my own and am now forced to ask for help.

I followed the instructions here:
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows

I am using Strawberry Perl. I downloaded Boost 1.59.0 per the instructions. 

I ran into an early roadblock which I was able to get past by manually adding `BOOST_ROOT` and `BOOST_LIBRARYDIR` to my system variables (was not included in the above instructions). I rebooted after doing so.

Right now, when I try to build (`perl Build.PL` in the `slic3r` directory that was created after running `git clone git://github.com/alexrj/Slic3r.git`) it chugs along happily for a minute or two, then fails. The following text file is what I see.

[log.txt](https://github.com/alexrj/Slic3r/files/836123/log.txt)

Here is the mentioned log file, which doesn't seem to contain any further useful information.

[build.log.txt](https://github.com/alexrj/Slic3r/files/836124/build.log.txt)

It seems to be complaining about `boost/thread.hpp` not existing. However, I confirmed that `thread.hpp` exists in the `boost` subdirectory of the `boost_1_59_0` directory. I have `BOOST_ROOT` set to that `boost_1_59_0` directory. I have no idea why there is any difficulty finding `thread.hpp` or whether that is even the issue at all.

I tried re-running `b2 toolset=gcc link=static runtime-link=static variant=release` with the addition of `--with-system --with-thread` which was mentioned in a file I found in `slic3r\src` called `windows-build.txt` but it didn't make any difference. I see a lot of stuff mentioned in that `windows-build.txt` file that is not mentioned on the "Running Slic3r from git on Windows" page. Do I need to be installing `mingw` and `cmake` first for the instructions to work? I started to get the feeling I was heading in the wrong direction, which lead to me posting this.

Thank you.

alexrj, 2017-03-11:
I'm not able to answer the question, but did you notice we have automated builds right now? You can download a fresh compiled package from the repository home page here on GitHub.

reeseallen, 2017-03-11:
I did not see that. I see where I can manually download a zip of the source, but where is the option to do an automated build? I'm hoping to compile something newer than the over-a-year-old 1.2.9 windows release available on the slic3r website, in the hopes that the issue I have has been fixed.

lordofhyphens, 2017-03-11:
https://github.com/alexrj/Slic3r/blob/master/README.md Says right there.

lordofhyphens, 2017-03-11:
Your first log entry indicated that the path to the boost include dir isn't present.  The instructions on the wiki said to put it in C:\dev.

Either put the Boost 1.59.0 directory in C:\dev, or set BOOST_DIR, or set both BOOST_INCLUDEDIR and BOOST_LIBRARYDIR. Build.PL isn't quite smart enough to check that the headers can be found (just that you can link the libraries). 

Also, if you're using Strawberry Perl 5.18.1.4 (64-bit) or 5.24 (32-bit) you can use one of the archives linked below to bypass needing to install most of the perl-based dependencies. 
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Dependencies/Dependencies-1.3.0-dev#files

References: 
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#getting-the-perl-module-dependencies
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#installing-the-boost-libraries
https://github.com/alexrj/Slic3r/blob/master/xs/Build.PL#L49
https://github.com/alexrj/Slic3r/blob/master/xs/Build.PL#L51

closed 2017-03-11 by lordofhyphens