slic3r/Slic3r #3755: 1.2.9 Spiral vase unexpected extrusion in place of travel move

motorb, 2017-03-12:
On moving from bottom layers infill to spiral vase start point, the gcode generated is an extrusion instead of a travel move. Screenshot etc attached

[image]
.

1.2.9 paired with Repetier host 1.6.2, Windows 7

lordofhyphens, 2017-03-12:
@motorb 
Please read and edit your issue to provide the requested information:
https://github.com/alexrj/Slic3r/blob/master/.github/CONTRIBUTING.md

I can't tell what's going on with what you provided.

Then try again with https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-dev-a193297#files to ensure that it's still happening with current master (you should be using current master anyway).

motorb, 2017-03-12:
Apologies, for some reason github isn't accepting the .zip file with the stl and configuration files.

motorb, 2017-03-12:
OK, so if I rename my zip file to gz it will upload. great. I'll edit the issue now.

lordofhyphens, 2017-03-12:
Don't forget to add a screenshot from Slic3r's plater too and check with 1.3.0-dev (linked above).

lordofhyphens, 2017-03-12:
That white line in your screenshot looks like a travel move to me, so I really don't know what you are seeing that is an issue.

motorb, 2017-03-13:
It's an extrusion move, highlighted as I've highlighted the Gcode line in the pane on the right. Further confirmed when I printed the part....

alexrj, 2017-03-13:
Yeah please, test with 1.3.0-dev.

alexrj, 2017-04-04:
@motorb, can you confirm this is fixed using a fresh build? You can download it at http://dl.slic3r.org/dev/

ruevs, 2018-09-26:
Version 1.3.1-dev
Build commit: d17659b67c65bb5003ec92bf0d56ed7f8810e970

Occasionally moves that should be travel moves are generated with extrusions.
Models similar to the attached one with "three spokes" show this on layers other than the first one as well.
[image]
[Slic3rBug.zip](https://github.com/slic3r/Slic3r/files/2419890/Slic3rBug.zip)

ruevs, 2018-09-26:
Bu the way this is a regression of sorts. For example Slic3r-master.2018.04.26.1669.8a93fbb.32bit.zip (which I had laying around) does NOT have this problem when slicing the exact same object.
[image]