slic3r/Slic3r #3756: Bug: Multiple Extruders/hotends - Wrong extruders temperature

Remcatfoil, 2017-03-12:
### Version
Slic3r 1.2.9

### Operating system type + version
Win10

### Behavior
I built a TWO hotends 3d printer based on prusa model.
When I add a stl file, and i click "Setting..." extruder 1 (or defaut) then hotend temperature corresponds to the defined value in "filament settings", here 180°C

_; generated by Slic3r 1.2.9 on 2017-03-12 at 10:26:33

; external perimeters extrusion width = 0.35mm
; perimeters extrusion width = 0.44mm
; infill extrusion width = 0.44mm
; solid infill extrusion width = 0.44mm
; top infill extrusion width = 0.44mm

M107
M190 S57 ; set bed temperature
M104 **S180** ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

M109 **S180** ; wait for temperature to be reached_

When i choose setting Extruder 2, then hotend temp is 200°C.*

_; generated by Slic3r 1.2.9 on 2017-03-12 at 10:26:11

; external perimeters extrusion width = 0.35mm
; perimeters extrusion width = 0.44mm
; infill extrusion width = 0.44mm
; solid infill extrusion width = 0.44mm
; top infill extrusion width = 0.44mm

M107
M190 S57 ; set bed temperature
M104 **S200** T1 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

M109 **S200** T1 ; wait for temperature to be reached_

Same behavior if i select in printer setting tab, Capabilities Extruders 2.
I can not defined temperature for the second hotend, and temperature of Extruder2 is the default value at 200°C.

[config.zip](https://github.com/alexrj/Slic3r/files/836412/config.zip)

Remy

lordofhyphens, 2017-03-12:
Each extruder gets assigned a filament type, and the default filament
temperature setting is 200C.

Please check 1.3.0-dev (see README.MD for links) and see if this is still a
problem. If it is I think we can probably only heat up the used extruders
in the print.

Remcatfoil, 2017-03-12:
Thank you for your prompt answer.
I checked 1.3.0-dev. Same issue, extruder 2 is at default temp, i.e. 200C.
Best,
Remy

lordofhyphens, 2017-03-12:
Why does your config.ini say it was generated by 1.1.7 last December?

Also, when you have 2 extruders defined, the Plater then shows two profiles for filament. What is the temperature listed for the second one selected? That's the filament temperature for extruder 2.

lordofhyphens, 2017-03-12:
A screenshot of your Slic3r plater window interface would be useful here.

Remcatfoil, 2017-03-12:
Ok, i understand now. I need two differents config files.
You are right i see the profile for the two filaments in the plater tab. Then i choose each config file that defines each filament..
When i export g-code i obtain the correct temperature for each filaments.
Thank you for your help, this solves the pb.

closed 2017-03-12 by lordofhyphens

Remcatfoil, 2017-03-12:
Quite tricky to understand how the config files/presets behaves. For ex. I do not understand the role of the " forbidden" sign which should remove a config preset. I seems not to remove a preset: when i restart the soft there is persistence of the removed preset.
Also, what is a difference between an ini file that could be exported, and a preset that could be saved with the "disk" icon? Where the presets are saved on the hard drive, since if i reinstall the soft it keeps in memory the previous presets? In the registry?
Thank you for you help,
Remy

lordofhyphens, 2017-03-12:
Switch to Expert mode.
All configurations are saved on the hard drive (assuming you save them).

On Windows it is in your profile dir. On OSX or Linux they are in ~/.Slic3r