slic3r/Slic3r #3757: Feature request: avoid abrupt speed-up around the small perimeter threshold

DrLex0, 2017-03-12:
### Version
1.2.9

### Operating system type + version
Mac OS X 10.11.6

### Request
* It makes sense to set small perimeters speed much lower than regular perimeters speed to improve accuracy. However, when printing with multiple perimeters, the small perimeter threshold is applied to each of the perimeters independently even if they belong to the same contour. For instance, when printing a simple cylinder with two perimeters, and the radius of the cylinder is just a tad larger than the small perimeter threshold, then the inner perimeter will be printed at small perimeter speed and the extruder will abruptly speed up to normal speed when starting the outer perimeter, which usually results in visible under-extrusion and ugly scars on the print.

I'm not sure what would be the best way to avoid this, here are two proposals:
* If one perimeter is printed at the small perimeter speed, print all other perimeters adjacent to it at the same speed. This could become complicated when considering non-trivial objects and could needlessly slow down the print.
* Get rid of the hard threshold, and instead use a linear transition zone between small perimeter speed and normal perimeter speed. This is easy to implement and should work pretty well.

bubnikv, 2017-03-13:
The correct solution IMHO is to have limits on the volumetric extrusion
rate and on the print head speed, and to smoothly vary the print speed to
achieve these limits using some form of a g-code post-processor. Ideally
the post-processor would know the firmware and it's acceleration behavior.
This should replace the current constant volumetric flow mode.

DrLex0, 2017-03-13:
That would be a very welcome feature indeed, but still it seems to make sense to gradually vary the speed for small perimeters instead of applying a hard threshold, because the effects of printing small perimeters too quickly will also disappear gradually with increasing diameter.