slic3r/Slic3r #3758: error during building. Last update

dobrzann, 2017-03-13:
### Version
1.2.9-718-g6e427028

### Operating system type + version
Windows 10 x64

### Behavior
error while building slic3r

cpanm (App::cpanminus) 1.7012 on perl 5.018004 built for MSWin32-x64-multi-thread
Work directory is C:\Users\ppp/.cpanm/work/1489387145.4576
You have make C:\strawberry\c\bin\dmake.exe
You have LWP 6.23
Falling back to Archive::Tar 2.02
Checking if you have ExtUtils::MakeMaker 6.58 ... Yes (7.24)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
--> Working on ./xs
Entering D:/Drukarka_3D/slic3r/xs
Checking if you have Module::Build 0.36 ... Yes (0.4210)
Configuring D:/Drukarka_3D/slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have Test::More 0 ... Yes (1.302078)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have Module::Build 0.3601 ... Yes (0.4210)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See C:\Users\ppp\.cpanm\work\1489387145.4576\build.log for details. Retry with --force to force install it.

Everything was fine, before last update.
What should I do?

lordofhyphens, 2017-03-13:
src\libslic3r\Config.cpp: In member function 'Slic3r::ConfigOptionDef* Slic3r::ConfigDef::add(const t_config_option_key&, const Slic3r::ConfigOptionDef&)':
src\libslic3r\Config.cpp:220:19: error: 'Slic3r::t_optiondef_map' has no member named 'emplace'

@alexrj broke something in a recent commit or we need to add a build flag

lordofhyphens, 2017-03-13:
I see what happened. Strawberry perl 5.18 uses gcc 4.7, which doesn't support the `emplace()` function. Joy.

I've updated the wiki instructions and provided a modified distribution that has a newer gcc. You will need to re-build Boost.

Links:
https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=strawberry-5.18-gcc-4.9.7z
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#strawberry-perl

closed 2017-03-13 by lordofhyphens