slic3r/Slic3r #3761: "No top_infill_pattern in ConfigOptionsGroup config at lib/Slic3r/GUI/OptionsGroup.pm line 369."

gege2b, 2017-03-14:
Hi all

Using the c93314c windows build available on @lordofhyphens 's bintray (https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-dev-c93314c ), I face this error when launching Slic3r

Seems related to commit 4b4a0c8a5cc7c607fee7680e7c5819d9a697c317 

### Version
c93314c 

### Operating system type + version
Windows 10 x64

### Behavior
mentionned error is shown in console windows, then quit.

lordofhyphens, 2017-03-14:
99% sure this is a build system problem, not a SW issue.

lordofhyphens, 2017-03-14:
Which I am working on.

born2b, 2017-03-14:
Hello.
Me, too.

Slic3r-master.2017.03.12.487.e655359 or older
I start it safely.

Slic3r-master.2017.03.12.489.6e42702 or newer
I get an error.

I am sorry that my English composition is not good.

lordofhyphens, 2017-03-14:
As I said, I know. There is a general build problem on Windows currently.

lordofhyphens, 2017-03-14:
Locking to avoid a plethora of me too posts.

lordofhyphens, 2017-03-14:
This is a Windows build issue. I am working in fixing it. Basically Strawberry Perl 5.18 ships a compiler that is too old.

lordofhyphens, 2017-03-14:
Build server is fixed with a workaround, I will be employing a more permanent fix when I have time.

closed 2017-03-14 by lordofhyphens