slic3r/Slic3r #3762: Not able to compile GUI

freiser77, 2017-03-14:
### Version
1.2.9-726-gc93314c

### Operating system type + version
macOS Sierra

### Behavior
Compile with ```perl Build.PL --gui``` fails with the following error message:
```
Configuring Wx-0.9928 ... N/A
! Configure failed for Wx-0.9928. See /Volumes/LaCie/florian/.cpanm/work/1489500213.51852/build.log for details.
! Installing the dependencies failed: Module 'Wx::build::MakeMaker' is not installed
! Bailing out the installation for Wx-GLCanvas-0.09.
Don't worry, this module is optional.
The following prerequisites failed to install: Wx
```

lordofhyphens, 2017-03-14:
This is black magic. You're unfortunately on your own and you'll need to dig through the CPANM logs to work out what is going on.

We have OSX builds:
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-dev-c20d44c

freiser77, 2017-03-14:
I know of this builds, but as I want to help migrating to C++, this does not help me.

Could you please help me to solve this problem, as I am not proficient in perl.
And please: do not close the issue without given me a chance to reply.

lordofhyphens, 2017-03-14:
Step 1: examine /Volumes/LaCie/florian/.cpanm/work/1489500213.51852/build.log
Step 2: Walk backwards through the dependencies and error messages. Google is your friend here.

Or... 

Ensure that you followed the instructions at https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X (I just updated them to include the perlbrew instructions). 

Using perlbrew is *highly* recommended (it's the setup the build server uses). Make sure you build with threads (https://colinnewell.wordpress.com/2012/04/24/perlbrew-with-options-like-threading/). The build server uses Perl 5.24.0. Once perl itself is built, switch to using its environment (`source ~/perl5/perlbrew/etc/bashrc && perlbrew switch 5.24.0` and install local::lib and App::cpanminus with cpan (`cpan local::lib App::cpanminus`).

From there, you should be able to check out Slic3r and run `perl Build.pl` and `perl Build.pl --gui` to install modules (don't forget to compile Boost and set BOOST_DIR to point to it). 

The rest of troubleshooting is just reading error messages and googling them.

If you want interactive help, it is probably better to head to #slic3r in freenode. I don't know who there (or in #reprap) who would have a Mac.

As I said, I'm not proficient in CPAN (which is really what is the problem here, some dependencies are failing to build).

lordofhyphens, 2017-03-14:
I know that if you're trying to build on a headless OSX system the Wx tests will fail.

alexrj, 2017-03-14:
@freiser77, come to #slic3r on FreeNode. I have OS X and I can (try to) help you. We're interested in your help for the C++ port.

lordofhyphens, 2017-03-15:
Would pulling local-lib out of the dmg work? Assuming the same major/minor version of Perl was used of course. 

That might let @freiser77 skip building all of the dependencies entirely (I'd still use Perlbrew).

Here's the Perl configuration summary off of the build server, by the way. Of particular note is the config-args entry, you should be able to pass them to perlbrew as  you see them (you can ignore minimum version for osx unless you too want to deal with making osx packages)
````
Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=darwin, osvers=13.4.0, archname=darwin-thread-multi-2level
    uname='darwin 3522.local 13.4.0 darwin kernel version 13.4.0: sun aug 17 19:50:11 pdt 2014; root:xnu-2422.115.4~1release_x86_64 x86_64 '
    config_args='-de -Dccflags=-mmacosx-version-min=10.7 -fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector -Dccdlflags=-mmacosx-version-min=10.7 -Dldflags=-mmacosx-version-min=10.7 -fstack-protector -Dlddlflags=-mmacosx-version-min=10.7 -bundle -undefined dynamic_lookup -fstack-protector -Dprefix=/Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7 -Duse64bitall -Dusethreads -Aeval:scriptdir=/Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/bin'
    hint=recommended, useposix=true, d_sigaction=define
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=define, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cc', ccflags ='-mmacosx-version-min=10.7 -fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector',
    optimize='-O3',
    cppflags='-arch x86_64 -mmacosx-version-min=10.7 -fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector'
    ccversion='', gccversion='4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)', gccosandvers=''
    intsize=4, longsize=8, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=define, longlongsize=8, d_longdbl=define, longdblsize=16, longdblkind=3
    ivtype='long', ivsize=8, nvtype='double', nvsize=8, Off_t='off_t', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='env MACOSX_DEPLOYMENT_TARGET=10.3 cc -arch x86_64', ldflags ='-mmacosx-version-min=10.7 -fstack-protector'
    libpth=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../lib/clang/6.0/lib /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib /usr/lib
    libs=-lpthread -ldbm -ldl -lm -lutil -lc
    perllibs=-lpthread -ldl -lm -lutil -lc
    libc=, so=dylib, useshrplib=false, libperl=libperl.a
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_dlopen.xs, dlext=bundle, d_dlsymun=undef, ccdlflags='-mmacosx-version-min=10.7'
    cccdlflags=' ', lddlflags='-mmacosx-version-min=10.7 -bundle -undefined dynamic_lookup -fstack-protector'

Characteristics of this binary (from libperl): 
  Compile-time options: HAS_TIMES MULTIPLICITY PERLIO_LAYERS
                        PERL_COPY_ON_WRITE PERL_DONT_CREATE_GVSV
                        PERL_HASH_FUNC_ONE_AT_A_TIME_HARD
                        PERL_IMPLICIT_CONTEXT PERL_MALLOC_WRAP
                        PERL_PRESERVE_IVUV USE_64_BIT_ALL USE_64_BIT_INT
                        USE_ITHREADS USE_LARGE_FILES USE_LOCALE
                        USE_LOCALE_COLLATE USE_LOCALE_CTYPE
                        USE_LOCALE_NUMERIC USE_LOCALE_TIME USE_PERLIO
                        USE_PERL_ATOF USE_REENTRANT_API
  Locally applied patches:
	Devel::PatchPerl 1.38
  Built under darwin
  Compiled at Mar  6 2017 03:21:46
  %ENV:
    PERLBREW_BASHRC_VERSION="0.78"
    PERLBREW_HOME="/Users/buildbot/.perlbrew"
    PERLBREW_MANPATH="/Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/man"
    PERLBREW_PATH="/Users/buildbot/perl5/perlbrew/bin:/Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/bin"
    PERLBREW_PERL="perl-5.24.0_osx10.7"
    PERLBREW_ROOT="/Users/buildbot/perl5/perlbrew"
    PERLBREW_VERSION="0.78"
  @INC:
    /Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/lib/site_perl/5.24.0/darwin-thread-multi-2level
    /Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/lib/site_perl/5.24.0
    /Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/lib/5.24.0/darwin-thread-multi-2level
    /Users/buildbot/perl5/perlbrew/perls/perl-5.24.0_osx10.7/lib/5.24.0
````

freiser77, 2017-03-15:
The command ```perlbrew install perl-5.24.0 -Dusethreads -Duselargefiles -Dcccdlflags=-fPIC -Doptimize=-O2 --as threaded-perl-5.24.0``` from @lordofhyphens setup instructions fails with
```
chmod 644 "HiRes.bs"
"../../miniperl" "-I../../lib" "-I../../lib" "../../lib/ExtUtils/xsubpp"  -typemap "../../lib/ExtUtils/typemap" -typemap "typemap"  HiRes.xs > HiRes.xsc && mv HiRes.xsc HiRes.c
cc -c   -fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -Wall -Werror=declaration-after-statement -Wextra -Wc++-compat -Wwrite-strings -Wthread-safety -O2   -DVERSION=\"1.9733\" -DXS_VERSION=\"1.9733\" -fPIC "-I../.."  -DTIME_HIRES_NANOSLEEP -DTIME_HIRES_CLOCK_GETTIME -DTIME_HIRES_CLOCK_GETRES -DTIME_HIRES_CLOCK_NANOSLEEP -DTIME_HIRES_CLOCK -DTIME_HIRES_STAT=1 -DATLEASTFIVEOHOHFIVE HiRes.c
HiRes.xs:945:15: error: use of undeclared identifier 'darwin_time_mutex'
  MUTEX_INIT(&darwin_time_mutex);
              ^
1 error generated.
make[1]: *** [HiRes.o] Error 1
cc -c   -fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -Wall -Werror=declaration-after-statement -Wextra -Wc++-compat -Wwrite-strings -Wthread-safety -O2   -DVERSION=\"1.9733\" -DXS_VERSION=\"1.9733\" -fPIC "-I../.."  -DTIME_HIRES_NANOSLEEP -DTIME_HIRES_CLOCK_GETTIME -DTIME_HIRES_CLOCK_GETRES -DTIME_HIRES_CLOCK_NANOSLEEP -DTIME_HIRES_CLOCK -DTIME_HIRES_STAT=1 -DATLEASTFIVEOHOHFIVE HiRes.c
HiRes.xs:945:15: error: use of undeclared identifier 'darwin_time_mutex'
  MUTEX_INIT(&darwin_time_mutex);
              ^
1 error generated.
make[1]: *** [HiRes.o] Error 1
Unsuccessful make(dist/Time-HiRes): code=512 at make_ext.pl line 569.
make: *** [lib/auto/Time/HiRes/HiRes.bundle] Error 25
##### Brew Failed #####
```

What shall I do?

lordofhyphens, 2017-03-15:
@freiser77 Google indicates:
https://github.com/Homebrew/homebrew-core/issues/1957

alexrj, 2017-03-15:
@freiser77, try this one instead:

```
PERLBREW_CONFIGURE_FLAGS='-de -Dccflags="-fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector" \
-Dldflags="-fstack-protector" \
-Dlddlflags="-bundle -undefined dynamic_lookup -fstack-protector"' \
perlbrew install perl-5.24.1 --thread --64all -j 2 --as threaded-perl-5.24.1 --force
```

lordofhyphens, 2017-03-15:
@freiser77 webchat.freenode.net if you don't want to set up a local irc client.

alexrj, 2017-03-15:
If my last command fails try also:

```
PERLBREW_CONFIGURE_FLAGS='-de -Dccflags="-mmacosx-version-min=10.12 -fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector" \
-Dccdlflags="-mmacosx-version-min=10.12" \
-Dldflags="-mmacosx-version-min=10.12 -fstack-protector" \
-Dlddlflags="-mmacosx-version-min=10.12 -bundle -undefined dynamic_lookup -fstack-protector"' \
perlbrew install perl-5.24.1 --thread --64all -j 2 --as threaded-perl-5.24.1 --force
```

lordofhyphens, 2017-03-16:
I've updated the Wiki

alexrj, 2017-03-16:
@freiser77, are you able to compile the *xsgui* branch as well?
Keep us posted about that work, let's ensure we're on the same track.

freiser77, 2017-03-16:
@alexrj: I will continue to get it working this afternoon. Yesterday evening I had an error executing the ```... perl Build.PL``` part. But I do not have the error message handy right now. Will execute the build this afternoon again and then post the error messages I receive.

freiser77, 2017-03-16:
Now there are dependencies, which have to be installed by hand, like

- Devel::CheckLib
- ExtUtils::CppGuess

The error happening when ```cpanm -i ExtUtils::CppGuess``` is:
```

Configuring ExtUtils-CppGuess-0.11
Running Makefile.PL
Checking if your kit is complete...
Looks good
Generating a Unix-style Makefile
Writing Makefile for ExtUtils::CppGuess
Writing MYMETA.yml and MYMETA.json
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have File::Basename 0 ... Yes (2.85)
Checking if you have Data::Dumper 0 ... Yes (2.167)
Checking if you have Module::Build 0 ... Yes (0.4220)
Checking if you have Cwd 0 ... Yes (3.67)
Checking if you have ExtUtils::Manifest 0 ... Yes (1.70)
Checking if you have ExtUtils::MakeMaker 0 ... Yes (7.24)
Checking if you have Fatal 0 ... Yes (2.29)
Checking if you have Capture::Tiny 0 ... Yes (0.46)
Checking if you have File::Spec 0 ... Yes (3.67)
Checking if you have Test::More 0 ... Yes (1.302073)
Building and testing ExtUtils-CppGuess-0.11
cp lib/ExtUtils/CppGuess.pm blib/lib/ExtUtils/CppGuess.pm
Manifying 1 pod document
PERL_DL_NONLAZY=1 "/Volumes/LaCie/florian/perl5/perlbrew/perls/threaded-perl-blead/bin/perl5.25.11" "-MExtUtils::Command::MM" "-MTest::Harness" "-e" "undef *Test::Harness::Switches; test_harness(0, 'blib/lib', 'blib/arch')" t/*.t
# EUMM:$VAR1 = {
#           'CCFLAGS' => ' -mmacosx-version-min=10.12 -fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector -fstack-protector-strong -I/usr/local/include  -xc++ ',
#           'dynamic_lib' => {
#                              'OTHERLDFLAGS' => ' -lstdc++ '
#                            }
#         };
# ---
# MB:$VAR1 = {
#           'extra_compiler_flags' => ' -mmacosx-version-min=10.12 -fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector -fstack-protector-strong -I/usr/local/include  -xc++ ',
#           'extra_linker_flags' => ' -lstdc++ '
#         };
t/001_load.t .......... ok
Can't locate t/lib/TestUtils.pm in @INC (you may need to install the t::lib::TestUtils module) (@INC contains: /Volumes/LaCie/florian/.cpanm/work/1489670148.14432/ExtUtils-CppGuess-0.11/blib/lib /Volumes/LaCie/florian/.cpanm/work/1489670148.14432/ExtUtils-CppGuess-0.11/blib/arch /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11 /Volumes/LaCie/florian/perl5/lib/perl5/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11 /Volumes/LaCie/florian/perl5/lib/perl5/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5 /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11 /Volumes/LaCie/florian/perl5/lib/perl5/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5 /Volumes/LaCie/florian/perl5/perlbrew/perls/threaded-perl-blead/lib/site_perl/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/perlbrew/perls/threaded-perl-blead/lib/site_perl/5.25.11 /Volumes/LaCie/florian/perl5/perlbrew/perls/threaded-perl-blead/lib/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/perlbrew/perls/threaded-perl-blead/lib/5.25.11) at t/010_module_build.t line 5.
BEGIN failed--compilation aborted at t/010_module_build.t line 5.
# Looks like your test exited with 2 before it could output anything.
t/010_module_build.t .. 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 1/1 subtests 
Can't locate t/lib/TestUtils.pm in @INC (you may need to install the t::lib::TestUtils module) (@INC contains: /Volumes/LaCie/florian/.cpanm/work/1489670148.14432/ExtUtils-CppGuess-0.11/blib/lib /Volumes/LaCie/florian/.cpanm/work/1489670148.14432/ExtUtils-CppGuess-0.11/blib/arch /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11 /Volumes/LaCie/florian/perl5/lib/perl5/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11 /Volumes/LaCie/florian/perl5/lib/perl5/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5 /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5/5.25.11 /Volumes/LaCie/florian/perl5/lib/perl5/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/lib/perl5 /Volumes/LaCie/florian/perl5/perlbrew/perls/threaded-perl-blead/lib/site_perl/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/perlbrew/perls/threaded-perl-blead/lib/site_perl/5.25.11 /Volumes/LaCie/florian/perl5/perlbrew/perls/threaded-perl-blead/lib/5.25.11/darwin-thread-multi-2level /Volumes/LaCie/florian/perl5/perlbrew/perls/threaded-perl-blead/lib/5.25.11) at t/011_makemaker.t line 5.
BEGIN failed--compilation aborted at t/011_makemaker.t line 5.
# Looks like your test exited with 2 before it could output anything.
t/011_makemaker.t ..... 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 1/1 subtests 

Test Summary Report
-------------------
t/010_module_build.t (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 1 tests but ran 0.
t/011_makemaker.t   (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 1 tests but ran 0.
Files=3, Tests=2,  0 wallclock secs ( 0.01 usr  0.00 sys +  0.12 cusr  0.02 csys =  0.15 CPU)
Result: FAIL
Failed 2/3 test programs. 0/2 subtests failed.
make: *** [test_dynamic] Error 2
-> FAIL Installing ExtUtils::CppGuess failed. See /Volumes/LaCie/florian/.cpanm/work/1489670148.14432/build.log for details. Retry with --force to force install it.
```

lordofhyphens, 2017-03-16:
Error message says it can't find a utility file that should be in cpputil's
test folder.

I have never seen this error before. If you go into the cpan working
directory you can run the tests manually with prove (and check the status
of that file).

If the tests do pass, install with --force.

freiser77, 2017-03-16:
Okay, I could proceed. Now I receive the following error:
```
cpanm (App::cpanminus) 1.7042 on perl 5.025011 built for darwin-thread-multi-2level
Work directory is /Volumes/LaCie/florian/.cpanm/work/1489685262.20840
You have make /usr/bin/make
You have /usr/bin/curl
You have /usr/bin/tar: bsdtar 2.8.3 - libarchive 2.8.3
You have /usr/bin/unzip
Searching Alien::wxWidgets () on cpanmetadb ...
--> Working on Alien::wxWidgets
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Alien-wxWidgets-0.67.tar.gz
-> OK
Unpacking Alien-wxWidgets-0.67.tar.gz
Entering Alien-wxWidgets-0.67
Checking configure dependencies from META.json
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Checking if you have Module::Build 0.38 ... Yes (0.4220)
Configuring Alien-wxWidgets-0.67
Running Build.PL
do "patches/data-3.0.2" failed, '.' is no longer in @INC at Build.PL line 225.
Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Alien-wxWidgets' version '0.67'
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::CBuilder 0.24 ... Yes (0.280225)
Checking if you have File::Spec 1.50 ... Yes (3.67)
Checking if you have Module::Build 0.28 ... Yes (0.4220)
Checking if you have Module::Pluggable 2.6 ... Yes (5.2)
Building and testing Alien-wxWidgets-0.67
Can't use an undefined value as a HASH reference at inc/My/Build/Base.pm line 299.
Building Alien-wxWidgets
-> FAIL Installing Alien::wxWidgets failed. See /Volumes/LaCie/florian/.cpanm/work/1489685262.20840/build.log for details. Retry with --force to force install it.
```

lordofhyphens, 2017-03-16:
Probably related (found via searching for `do patches/data-3.0.2 failed wxperl` on Google): http://stackoverflow.com/questions/32689670/alienwxwidgets-install-fails-on-osx10-10

freiser77, 2017-03-16:
Thanks for the tip. But it does not help. When I install wxmac with homebrew, it's not working, too.

lordofhyphens, 2017-03-16:
Yep, that's why dependencies are hell.
It's a long shot, but the build server dependency package *may* work, although you'll probably have to rewrite the lib paths (see package/osx/make_dmg.sh). At least if the new osx packaged Slic3r works, this may work as well. Extract into your Slic3r directory and consult how the dylibs are modified in package/osx/make_dmg.sh.

https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-perl-dependencies-5.24.0-osx107.bz2

lordofhyphens, 2017-03-16:
Getting Alien::wxWidgets to build is probably one of the most painful dependencies to build.

lordofhyphens, 2017-03-16:
https://github.com/alexrj/Slic3r/blob/master/package/osx/make_dmg.sh#L91 Specifically, if you copied those few lines out into a shell script and set $macosfolder to your Slic3r checkout path it may just work.

freiser77, 2017-03-17:
@lordofhyphens: The bz2 file only contains one file. Is that correct?

lordofhyphens, 2017-03-17:
It is a bzip'd tarball of a directory.

tar -xvjf /path/to/file.tar.bz2

freiser77, 2017-03-17:
```
Florians-iMac:Slic3r florian$ BOOST_DIR=~/boost_1_59 SLIC3R_STATIC=1 perl Build.PL
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
Don't worry, this module is optional.
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
Don't worry, this module is optional.
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
Don't worry, this module is optional.
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
Don't worry, this module is optional.
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
Don't worry, this module is optional.
ListUtil.c: loadable library and perl binaries are mismatched (got handshake key 0xd880080, needed 0xdb00080)
The XS/C++ code failed to compile, aborting
```

lordofhyphens, 2017-03-17:
Darn. We'd probably need to get you a copy of our perl (which is compiled
for compatibility) too for that archive to work.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2017-03-23:
@freiser77, no success yet?
I have now access to a Sierra machine, so I can help you if you need.

alexrj, 2017-03-23:
Relevant wxWidgets bug: http://trac.wxwidgets.org/ticket/17639
Note to self: solution seems to be compiling it with `--disable-qtkit --disable-mediactrl`.

freiser77, 2017-03-23:
@alexrj: No, no luck yet. I even have tried it by installing perl over macports. I'll try your suggestion on the weekend.

freiser77, 2017-03-25:
This is the current error message:

```
Florians-iMac:Slic3r florian$ WX_DIR=~/wxWidgets-3.1.0/ perl Build.PL --gui
App::cpanminus is up to date. (1.7042)
Class::Accessor is up to date. (0.34)
Growl::GNTP is up to date. (0.21)
LWP::UserAgent is up to date. (6.24)
Net::Bonjour is up to date. (0.96)
OpenGL is up to date. (0.70)
Socket is up to date. (2.024)
--> Working on Wx
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9928.tar.gz ... OK
==> Found dependencies: Alien::wxWidgets
--> Working on Alien::wxWidgets
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Alien-wxWidgets-0.67.tar.gz ... OK
Configuring Alien-wxWidgets-0.67 ... OK
Building and testing Alien-wxWidgets-0.67 ... FAIL
! Installing Alien::wxWidgets failed. See /Volumes/LaCie/florian/.cpanm/work/1490467645.2472/build.log for details. Retry with --force to force install it.
! Installing the dependencies failed: Module 'Alien::wxWidgets' is not installed
! Bailing out the installation for Wx-0.9928.
--> Working on Wx::GLCanvas
Fetching http://www.cpan.org/authors/id/M/MB/MBARBON/Wx-GLCanvas-0.09.tar.gz ... OK
==> Found dependencies: Wx::build::MakeMaker
--> Working on Wx::build::MakeMaker
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9928.tar.gz ... OK
==> Found dependencies: Alien::wxWidgets
--> Working on Alien::wxWidgets
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Alien-wxWidgets-0.67.tar.gz ... OK
Configuring Alien-wxWidgets-0.67 ... OK
Building and testing Alien-wxWidgets-0.67 ... FAIL
! Installing Alien::wxWidgets failed. See /Volumes/LaCie/florian/.cpanm/work/1490467650.2492/build.log for details. Retry with --force to force install it.
! Installing the dependencies failed: Module 'Alien::wxWidgets' is not installed
! Bailing out the installation for Wx-0.9928.
! Installing the dependencies failed: Module 'Wx::build::MakeMaker' is not installed
! Bailing out the installation for Wx-GLCanvas-0.09.
Don't worry, this module is optional.
The following prerequisites failed to install: Wx
```

Is there any possibility to direct the build to the wxWidgets-dir?
Mine is installed in ```~/wxWidgets-3.1.0```

alexrj, 2017-03-26:
@freiser77, I started installing stuff on the Sierra machine I have access to. I'm not there full time so it might take several days, but I'll keep you updated with my findings and I will update the wiki page.

alexrj, 2017-03-26:
I did progress. I'm close but I still need to finish a patch for wxWidgets.

alexrj, 2017-03-30:
I think I sorted it out. The wiki page is updated with the instructions:
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X

I won't be able to continue before Sunday or Monday, so @freiser77 if you want to try in these days let me know how it goes.

freiser77, 2017-03-30:
@alexrj: As macOs does not have the openssl headers, I had to follow the advice in this article: https://medium.com/@deepak.gulati/installing-net-ssleay-on-os-x-el-capitan-446c40cffeba

Only version 0.98zg was functioning.
After that, I've installed Net::SSLeay and LWP-Protocol-https-6.07 by

```
cpanm --locla-lib local-lib/ Net::SSLeay
cpanm --local-lib local-lib/ LWP::Protocol::https --force
```

After having done that, the build failed, because the patch  file patch-wxwidgets-noquicktime.diff was not available. Where did you find it? Maybe we should include it in git?

lordofhyphens, 2017-03-30:
@freiser77 

I suspect it's derived from http://trac.wxwidgets.org/changeset/817bd0076f61ed96694ec6cebba46ca5df495998/git-wxWidgets 

Reference: http://trac.wxwidgets.org/ticket/17639

Another possible option is to try to build with 3.1.0 instead (which should have that commit in it). I have a hacked version of Alien::wxWidgets and wxperl for this, but it is untested on OSX and I don't have an OSX system to test with. 

https://github.com/lordofhyphens/alien-wxwidgets-3.1.0
https://github.com/lordofhyphens/wxperl-3.1.0

Brief docs (what I had done on the Win10 system I was working with): https://github.com/alexrj/Slic3r/wiki/Building-wxWidgets-3.1.0

freiser77, 2017-04-01:
@alexrj: I've tried to use your patched versions. But the following error shows:

```
Florians-iMac:alien-wxwidgets-3.1.0 florian$ perl Build.PL --wxWidgets-build=1 --wxWidgets-version=3.1.0
Which archive type? [tar.bz2 ]
tar.bz2 
do "patches/data-3.1.0" failed, '.' is no longer in @INC at Build.PL line 226, <STDIN> line 1.
Do you want to include OpenGL support [yes ]
yes 
Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Alien-wxWidgets' version '0.67'
Florians-iMac:alien-wxwidgets-3.1.0 florian$ perl Build
Building Alien-wxWidgets
Can't use an undefined value as a HASH reference at inc/My/Build/Base.pm line 299.
```

alexrj, 2017-04-03:
The patch is available at https://gist.githubusercontent.com/alexrj/c2de82c05f6006b49c5029fc78bcaa87/raw/2164a23f6d580fe31b995bfcd17292e0cf4a87f8/patch-wxwidgets-noquicktime.diff
I included it in the wiki now.

Regarding your last message, you issued a command which is not included in the OSX wiki guide. You probably took it from the Windows guide.

alexrj, 2017-04-03:
I confirm the instructions in https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X work successfully on Sierra now.

I suggest to clean everything and follow them step by step.

freiser77, 2017-04-04:
@alexrj : I've cleaned everything. But the patching 

```
PERL_USE_UNSAFE_INC=1 CXX="clang++ -mmacosx-version-min=10.12" CPPFLAGS="-mmacosx-version-min=10.12" LDFLAGS="-mmacosx-version-min=10.12" cpanm --local-lib local-lib Alien::wxWidgets -v --reinstall --configure-args="--wxWidgets-build=1 --wxWidgets-extraflags=\"--with-macosx-version-min=10.12 --disable-qtkit --disable-mediactrl --disable-webkit --disable-webview\" --wxWidgets-userpatch=$(pwd)/patch-wxwidgets-noquicktime.diff"
```

 of wxWidgets is not functioning.
The error when running it in ~/Slic3r is :

```
Florians-iMac:Slic3r florian$ PERL_USE_UNSAFE_INC=1 CXX="clang++ -mmacosx-version-min=10.12" CPPFLAGS="-mmacosx-version-min=10.12" LDFLAGS="-mmacosx-version-min=10.12" cpanm --local-lib local-lib Alien::wxWidgets -v --reinstall --configure-args="--wxWidgets-build=1 --wxWidgets-extraflags=\"--with-macosx-version-min=10.12 --disable-qtkit --disable-mediactrl --disable-webkit --disable-webview\" --wxWidgets-userpatch=$(pwd)/patch-wxwidgets-noquicktime.diff"
cpanm (App::cpanminus) 1.7043 on perl 5.026000 built for darwin-thread-multi-2level
Work directory is /Volumes/LaCie/florian/.cpanm/work/1491288082.68662
You have make /usr/bin/make
You have LWP 6.25
You have /usr/bin/tar: bsdtar 2.8.3 - libarchive 2.8.3
You have /usr/bin/unzip
Searching Alien::wxWidgets () on cpanmetadb ...
--> Working on Alien::wxWidgets
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Alien-wxWidgets-0.67.tar.gz ... OK
Unpacking Alien-wxWidgets-0.67.tar.gz
x Alien-wxWidgets-0.67
x Alien-wxWidgets-0.67/Build.PL
x Alien-wxWidgets-0.67/Changes
x Alien-wxWidgets-0.67/Makefile.PL
x Alien-wxWidgets-0.67/MANIFEST
x Alien-wxWidgets-0.67/META.json
x Alien-wxWidgets-0.67/META.yml
x Alien-wxWidgets-0.67/README.txt
x Alien-wxWidgets-0.67/inc
x Alien-wxWidgets-0.67/inc/latest.pm
x Alien-wxWidgets-0.67/inc/bin
x Alien-wxWidgets-0.67/inc/bin/patch
x Alien-wxWidgets-0.67/inc/bin/patch.exe
x Alien-wxWidgets-0.67/inc/inc_Archive-Extract
x Alien-wxWidgets-0.67/inc/inc_Archive-Extract/Archive
x Alien-wxWidgets-0.67/inc/inc_Archive-Extract/Archive/Extract.pm
x Alien-wxWidgets-0.67/inc/inc_File-Fetch
x Alien-wxWidgets-0.67/inc/inc_File-Fetch/File
x Alien-wxWidgets-0.67/inc/inc_File-Fetch/File/Fetch.pm
x Alien-wxWidgets-0.67/inc/inc_IPC-Cmd
x Alien-wxWidgets-0.67/inc/inc_IPC-Cmd/IPC
x Alien-wxWidgets-0.67/inc/inc_IPC-Cmd/IPC/Cmd.pm
x Alien-wxWidgets-0.67/inc/inc_Locale-Maketext-Simple
x Alien-wxWidgets-0.67/inc/inc_Locale-Maketext-Simple/Locale
x Alien-wxWidgets-0.67/inc/inc_Locale-Maketext-Simple/Locale/Maketext
x Alien-wxWidgets-0.67/inc/inc_Locale-Maketext-Simple/Locale/Maketext/Simple.pm
x Alien-wxWidgets-0.67/inc/inc_Module-Load
x Alien-wxWidgets-0.67/inc/inc_Module-Load/Module
x Alien-wxWidgets-0.67/inc/inc_Module-Load/Module/Load.pm
x Alien-wxWidgets-0.67/inc/inc_Module-Load-Conditional
x Alien-wxWidgets-0.67/inc/inc_Module-Load-Conditional/Module
x Alien-wxWidgets-0.67/inc/inc_Module-Load-Conditional/Module/Load
x Alien-wxWidgets-0.67/inc/inc_Module-Load-Conditional/Module/Load/Conditional.pm
x Alien-wxWidgets-0.67/inc/inc_Params-Check
x Alien-wxWidgets-0.67/inc/inc_Params-Check/Params
x Alien-wxWidgets-0.67/inc/inc_Params-Check/Params/Check.pm
x Alien-wxWidgets-0.67/inc/inc_version
x Alien-wxWidgets-0.67/inc/inc_version/version.pm
x Alien-wxWidgets-0.67/inc/inc_version/version
x Alien-wxWidgets-0.67/inc/inc_version/version/vpp.pm
x Alien-wxWidgets-0.67/inc/latest
x Alien-wxWidgets-0.67/inc/latest/private.pm
x Alien-wxWidgets-0.67/inc/My
x Alien-wxWidgets-0.67/inc/My/Build.pm
x Alien-wxWidgets-0.67/inc/My/Build
x Alien-wxWidgets-0.67/inc/My/Build/Any_wx_config.pm
x Alien-wxWidgets-0.67/inc/My/Build/Any_wx_config_Bakefile.pm
x Alien-wxWidgets-0.67/inc/My/Build/Base.pm
x Alien-wxWidgets-0.67/inc/My/Build/gmake.mak
[... 138 lines cut ...]
|
|-#if wxOSX_USE_COCOA_OR_CARBON
|-    #include <QuickTime/QuickTime.h>
|-#endif
|+
|+
|+
|
| // ----------------------------------------------------------------------------
| // wxDataFormat
|--- src/osx/core/bitmap.cpp
|+++ src/osx/core/bitmap.cpp
--------------------------
Patching file src/osx/core/bitmap.cpp using Plan A...
Short hunk ignored.
1 out of 1 hunks ignored--saving rejects to src/osx/core/bitmap.cpp.rej

done
Error: 512 at inc/My/Build/Base.pm line 345.
FAIL
! Installing Alien::wxWidgets failed. See /Volumes/LaCie/florian/.cpanm/work/1491288082.68662/build.log for details. Retry with --force to force install it.
Florians-iMac:Slic3r florian$ 
```

Can you help me?

mkdryden, 2017-04-25:
@freiser77 I had the same issue. Remaking the patch file seemed to fix it: (change the extension from .txt to .diff)
[patch-wxwidgets-noquicktime.txt](https://github.com/alexrj/Slic3r/files/956754/patch-wxwidgets-noquicktime.txt)

ankang1943, 2017-05-03:
@mkdryden 

Thank you! you saved me!

alexrj, 2017-05-04:
@freiser77, did you have a chance to try @mkdryden's patch?
(I updated the one in the wiki to reflect changes proposed by him although I haven't tested them.)

alexrj, 2017-05-23:
I just re-tested the wiki procedure on a fresh Sierra machine and I confirm it works.
In the mean time, the wxPerl maintainer fixed compatibility with Sierra so my forks and the patch are no longer necessary.
@freiser77, looking forward for your help!

closed 2017-05-23 by alexrj