slic3r/Slic3r #3763: Slic3R crash in a directory

Pydo44, 2017-03-14:
### Version
last one 1,2,9

### Operating system type + version
Mac OS X 10.11.5

### Behavior
I put Slicer in the application directory and in Sub directory* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2017-03-15:
Try the current builds (look at readme.MD for link)

alexrj, 2017-03-15:
@Pydo44, can you confirm the current 1.3.0-dev builds fix the issue?

Pydo44, 2017-03-16:
Soory ... but i can found the readme.MD
where i can find it ?

lordofhyphens, 2017-03-16:
@Pydo44 No, I will not directly link to the newest version at the time of this writing.

1. Click the link that is clearly shown at the top of [Readme.MD](https://github.com/alexrj/Slic3r/blob/master/README.md). 
1. From there, you will see a list of versions, in order from newest to oldest. Click the top one.
1. Click "Files". 
1. Pick the file corresponding to the OSX installer (OSX installers are dmg files).

Pydo44, 2017-03-17:
hi
the same problem appear
where i can put the failure report

Pydo44, 2017-03-17:
@lordofhyphens i send you a mail with the report

lordofhyphens, 2017-03-17:
You can upload .txt files using the github Web interface. Do not directly
email me (or really anyone else) without being requested to ahead of time.

lordofhyphens, 2017-03-17:
[image]

Pydo44, 2017-03-17:
Sure but i don't want put on public web site my private informations

lordofhyphens, 2017-03-17:
@Pydo44 I also don't have a Mac. Strip out what looks "private", zip it up and post.

Pydo44, 2017-03-18:
i don't know what is private ....... so .......

lordofhyphens, 2017-03-18:
https://www.quora.com/Applications-on-my-Mac-crash-often-of-which-a-report-is-sent-to-Apple-Why-do-I-never-hear-back-or-see-updates-of-the-app-to-prevent-such-crashes 
10th entry from https://www.google.com/?q=what+is+private+information+in+an+osx+crash+report

alexrj, 2017-04-04:
@Pydo44, I was able to reproduce the issue. It's caused by the non-ASCII characters in the directory name (the é in Modélisation).  As a workaround, move it to some other place. I'm working on a fix.

alexrj, 2017-04-04:
By the way, there's nothing private in a crash dump.

alexrj, 2017-04-04:
@Pydo44, this should be fixed now.
Can you please download the most recent built at http://dl.slic3r.org/dev/mac/ and confirm it works now?

lordofhyphens, 2017-04-11:
@Pydo44 Please check to make sure that it is working for you so we can close this issue.

lordofhyphens, 2017-04-11:
@alexrj was there anything related to #2955 in it?

alexrj, 2017-05-01:
Yes @lordofhyphens, it was caused by the "Modélisation" directory in application path.

@Pydo44, can you please download the most recent built at http://dl.slic3r.org/dev/mac/ and confirm it works now? Thank you in advance!

lordofhyphens, 2017-05-05:
````
<keyno> i tried it in /User/me/täst 
<keyno> crashed
````

lordofhyphens, 2017-05-05:
````
<jammi> LoH|Lenneth: I tested with slic3r in various directories, even one named "täst" now and no issues whatsoever
<LoH|Lenneth> jammi: OSX? New build?
<LoH|Lenneth> Weird
<jammi> LoH|Lenneth: yes macos sierra
````
So now I'm confused.

lordofhyphens, 2017-05-05:
Also can't reproduce on Windows 10 x64. 

keyno from IRC indicates that after a reboot it is not reproducible, so probably fixed on OSX Sierra. 
jammi concurs.

closed 2017-05-05 by lordofhyphens