slic3r/Slic3r #3767: [1.3.0-dev] Failure to connect to Monoprice Select Mini printer on Linux @115200 baud

lordofhyphens, 2017-03-16:
### Version
c20d44c3887c39eaa6850470ba6571d7c92ab336

### Operating system type + version
Debian Linux 8.6 (Jessie), built from source.

### Behavior
Interface pauses for approximately 6-7 seconds and then errors out. 
I have successfully connected to this printer with Printrun at the same baud rate and port. 

[image]

alexrj, 2017-03-16:
I'm rewriting the serial code for Linux, it fails for me as well.

alexrj, 2017-03-25:
Fixed.

lordofhyphens, 2017-03-25:
2fc8152e24075dae96e09283fb548c2d82924fe2 Still fails here on my system. Printrun can connect.

alexrj, 2017-03-26:
Could you copypaste the console log from Printrun?
Also, if you have a change, could you enable the DEBUG_SERIAL in GCodeSender and attach its contents?

lordofhyphens, 2017-03-26:
There was only "DISCONNECTED" in the log last I tried, will check for new
commits and enable.

As far as I can tell this printer sends nothing on connection.

I was messing around with PUTTY and notices that if rts/cts flow control
was enabled then I could not send data (at least at 9600 baud). No flow or
xon/xoff worked iirc.

This broadcasts itself as a ttyACM device.

lordofhyphens, 2017-03-26:
From Printrun:
````
Connecting...
SENT: M105
RECV: ok N-1 P15 B15 T:17.5 /0.0 B:16.1 /0.0 T0:17.5 /0.0 @:0 B@:0
Printer is now online.
````

I disabled the monitoring function of Printrun.

lordofhyphens, 2017-03-26:
Works now.

closed 2017-03-26 by lordofhyphens