slic3r/Slic3r #3768: "Scale to size -> Uniformly" shows random result

gege2b, 2017-03-16:
### Version
_Version of Slic3r used goes here_
Current Win32 build from bintray  ( commit cdf48ad  )

### Operating system type + version
_What OS are you using, and state any version #s_
Win10 x64

### Behavior
When using "scale to size -> uniformly" function, it correctly resize the object the first time, but using it again makes the object size being completely random (but proportions are kept)

 Steps needed to reproduce the problem_
- Load an object (no matter witch one - [I used this one](http://www.thingiverse.com/thing:1871347) - max size is 82 in Y)
- Use "scale to size" -> "uniformly" to set the max. size to, say, 10.
Object is correctly scaled
- Use scale to size again, setting back the max. size to 82

Object is then 627,38x672,40x57,40 in size

* _Expected Results_
Object being resized to the good value

* _Actual Results_
values are way too far from what they should be

Tested with a lot of STLs and default config but here they are just in case

[config+stl.zip](https://github.com/alexrj/Slic3r/files/848522/config.stl.zip)

gege2b, 2017-03-16:
just tested under Linux Mint 18.1 and I got  the same issue

alexrj, 2017-03-18:
Whoops. Fixed!

closed 2017-03-18 by alexrj