slic3r/Slic3r #3769: Stops at "generating support material"

dobrzann, 2017-03-16:
### Version
cdf48ad

### Operating system type + version
Windows 10 x64

### Behavior
stops at "generating support material"

Just open .stl and wait for "generating suport material"

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs

[config and STL.zip](https://github.com/alexrj/Slic3r/files/848757/config.and.STL.zip)

dobrzann, 2017-03-16:
Problem was solved.

Incorrect value in "Overhang Threshold"

closed 2017-03-16 by dobrzann

lordofhyphens, 2017-03-16:
There's two meanings for overhang threshold now, sorry. Use a number for an angle (probably what happened here) and use a % for auto detection with that much overhang threshold relative to perimeter width.

The old behavior would be 200%

I have a new default to lessen that to more like 60% (perimeters that have more than 60% of their width over air get support).

Please experiment for your printer.