slic3r/Slic3r #3772: Can't locate auto/OpenGL/glGenBuffer.al in @INC (…) at /usr/lib64/perl5/vendor_perl/5.24.1/Slic3r/GUI/3DScene.pm line 861.

alexws54tk, 2017-03-17:
### Version
```
% slic3r.pl --version
1.3.0-dev
```
```
% git rev-parse HEAD                                         
cdf48ade2588e01c217e23391e356a580657ed47
```
```% git describe --tag
1.2.9-732-gcdf48ad
```
### Operating system type + version
* OS: Gentoo Linux
* Kernel: x86_64 Linux 4.1.15-1.1381_SKYN12nnmp
* DE: KDE5
  * KDE Frameworks: 5.29.0;
  * KDE plasma: 5.8.5;
  * Qt 5.6.2;

### Behavior
Slic3r don`t run in GUI mode
```
% slic3r.pl --gui

Can't locate auto/OpenGL/glGenBuffer.al in @INC (@INC contains:
 /usr/bin/local-lib/lib/perl5/5.24.1/x86_64-linux-thread-multi
 /usr/bin/local-lib/lib/perl5/5.24.1
 /usr/bin/local-lib/lib/perl5/x86_64-linux-thread-multi
 /usr/bin/local-lib/lib/perl5/5.24.0/5.24.0/x86_64-linux-thread-multi
 /usr/bin/local-lib/lib/perl5
 /usr/bin/lib
 /etc/perl
 /usr/local/lib64/perl5/5.24.1/x86_64-linux-thread-multi
 /usr/local/lib64/perl5/5.24.1
 /usr/lib64/perl5/vendor_perl/5.24.1/x86_64-linux-thread-multi
 /usr/lib64/perl5/vendor_perl/5.24.1
 /usr/local/lib64/perl5 /usr/lib64/perl5/vendor_perl/5.24.0
 /usr/lib64/perl5/vendor_perl
 /usr/lib64/perl5/5.24.1/x86_64-linux-thread-multi
 /usr/lib64/perl5/5.24.1 
.) at /usr/lib64/perl5/vendor_perl/5.24.1/Slic3r/GUI/3DScene.pm line 861.
```
In CLI mode run normally
```
% slic3r.pl \
 --layer-height 0.2 \
 --nozzle-diameter 0.4 \
 --print-center 0,0 \
 --gcode-flavor repetier \
 --use-relative-e-distances \
 --filament-diameter 1.75 \
 --temperature 216 \
 --first-layer-temperature 218 \
 --first-layer-speed 5 \
 --first-layer-height 0.3 \
 --solid-layers 4 \
 --avoid-crossing-perimeters \
 --raft-layers 2 \
 --retract-lift 1 \
 --wipe \
 --cooling \
 --slowdown-below-layer-time 12 \
 --fan-always-on \
 --extrusion-width 0.4 \
 fidgetwidget1piece.stl
=> Generating perimeters
=> Processing triangulated mesh
=> Infilling layers
=> Preparing infill
=> Generating support material
=> Weight: %.1fg, Cost: %.1f
=> Generating skirt
=> Generating brim
=> Exporting G-code to fidgetwidget1piece.gcode
Done. Process took 0 minutes and 0.783 seconds
Filament required: 1006.1mm (2.4cm3)
```

alexrj, 2017-03-17:
Can you try to git pull and retry now?

alexws54tk, 2017-03-20:
Hi friend.

I make this by hand from sources, and now it works.

alexrj, 2017-03-20:
Thank you for confirming!

closed 2017-03-20 by alexrj