slic3r/Slic3r #3773: Quick slice: Can't locate object method "extra_variables" via package "Slic3r::GUI::MainFrame

hroncok, 2017-03-17:
### Version
1.3.0-dev 4c78e54a4697c96333c71fa00fe39d7696373dcd (master)

`git describe --tag`: 1.2.9-743-g4c78e54

### Operating system type + version
Fedora 25 x86_64

### Behavior
* When I attempt to quick slice an STL, I get an error
* I've build Slic3r according to instructions [on wiki](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux)
* File | Quick Slice... (or Quick Slice Save As...)
* _Expected Results_:
   * dumps gcode somewhere next to the stl file
* _Actual Results_
  * `Can't locate object method "extra_variables" via package "Slic3r::GUI::MainFrame" at /home/vagrant/Slic3r/lib/Slic3r/GUI/MainFrame.pm line 420.`

#### STL/Config (.ZIP) where problem occurs
Any STL. You can try with [block.stl](https://github.com/admesh/python-admesh/blob/master/test/block.stl)

lordofhyphens, 2017-03-17:
Yeah, I can replicate this on c20d44c (windows) too. Suspect it's been broken for quite some time. I never actually use Quick Slice myself; what does this option do that the CLI doesn't?

hroncok, 2017-03-17:
I don't use it either, just wanted to reproduce #1636

alexrj, 2017-03-17:
Whoops, I always break Quick Slice. Fixed.

closed 2017-03-17 by alexrj

hroncok, 2017-03-17:
Confirmed.