slic3r/Slic3r #3778: Remove all croak() calls inside exception catch blocks.

lordofhyphens, 2017-03-20:
On newer versions of Perl, this causes the Perl interpreter to hang.

To avoid the problem, move the croak calls out of the exception handler and instead set a flag that is checked after the exception handler finishes.

bubnikv, 2017-03-20:
Did you try that? I believe this does not work. Once a C++ exception is
thrown, the call stack seems to be broken for good to be processed by the
Perl exception handler. A solution of course is to compile Perl with visual
studio. We are currently testing the VS builds for production and apart
from some ASCII stl import issues it seems to work just fine and 10% faster.

lordofhyphens, 2017-03-20:
I prefer to not include anything produced by MSVC in upstream.

alexrj, 2017-03-20:
We have no `croak()` calls in libslic3r, and libslic3r is already Perl-free (I designed it with this in mind). All the Perl-specific code is located in perlglue.cpp and xsinit.h and it's only enabled when the SLIC3RXS define is passed. Infact libslic3r as well as a limited slic3r executable can be already compiled with CMake (see the src/ directory).

The `croak()` calls are added by the ExtUtils::XSpp module which wraps all the method calls with this code:
```
   try {
      // ...
    }
    catch (std::exception& e) {
      croak("Caught C++ exception of type or derived from 'std::exception': %s", e.what());
    }
    catch (...) {
      croak("Caught C++ exception of unknown type");
    }
```

We could try what you are suggesting by implementing a custom C++ base exception for Slic3r and implement a custom exception handler as described in https://metacpan.org/pod/ExtUtils::XSpp::Exception, which would basically just return a string in RETVAL for those methods which are expected to throw something. I don't think that the Perl call stack is always compromised whenever an exception is caught, because we have some try/catch in libslic3r already; I think the bug only applies to the way `croak()` is implemented.

In order to handle the non-Slic3r exceptions I would redefine the `croak()` macro with a simple `printf(); exit(1)`. This can be done in xsinit.h which is included after the Perl headers and before bindings are generated.

lordofhyphens, 2017-03-28:
No longer necessary, "fixed" via using a gcc that supports SEH exceptions.

closed 2017-03-28 by lordofhyphens