slic3r/Slic3r #3779: Coursor jumps when entering values for move in parts modifier

foreachthing, 2017-03-20:
### Version
1.3.0-dev
slic3r-w32-exe.2017.03.02.380.b3434b7

### Operating system
Windows 7 Pro

### Behavior
In Settings of any Part:
* create modifier (I used a cylinder)
* add something to be modified (I used these two: Perimeter 5 and Infill 60%)
* try to move the modifier in XY or Z by entering values!
  * I tried to enter the value -58.4 - this is not possible! First of all, if you type 58 the cursor jumps around and you actually enter 85. You need to copy/paste such values.
* -58.4 comes out as 8.4-5

Expected Results:
* I can actually enter -58.4 in any of the textboxes for XY or Z

alexrj, 2017-03-20:
True, I can reproduce that. Thank you for reporting it.

alexrj, 2017-03-20:
Actually, you should share the full test case including the sizes of the object and the size of the modifier, otherwise I'm not sure I can reproduce it correctly.

alexrj, 2017-03-20:
I pushed a fix but I did not test it, can you please `git pull` and retry? Thank you.

foreachthing, 2017-03-21:
> can you please git pull and retry?
Sure. Done.
Digits work, decimal points do not. Decimal point only works when typed as first char. `0.` does not work. Or `12.3` doesn't work, etc.

alexrj, 2017-03-21:
I pushed one more fix. @foreachthing, could you please confirm this fixes the issue?

foreachthing, 2017-03-21:
- [x] ddad7ce works!

Thanks a lot!

lordofhyphens, 2017-03-22:
Thanks for confirming.

closed 2017-03-22 by lordofhyphens