slic3r/Slic3r #3780: Dos window full of warnings

cerdeira, 2017-03-20:
Version

1.3.0-dev slic3r-master.2017.03.20.570.9dbe2ee

Operating system

Windows 10

Behavior

When loading a config file, File/Load Config...  dos window gets full of warnings "Unknown Option..." blocks that repeat several times.

[image]

lordofhyphens, 2017-03-20:
All those warnings mean is that your print/filament/printer configs that aren't known to the version of Slic3r that you are using (either because the option was removed/replaced or you were using a fork that defined those that upstream doesn't).

closed 2017-03-20 by lordofhyphens

alexrj, 2017-03-20:
Yeah. None of those options are present in the original Slic3r.