slic3r/Slic3r #3781: Infill hols

cerdeira, 2017-03-20:
Version

1.3.0-dev slic3r-master.2017.03.20.570.9dbe2ee

Operating system

Windows 10

Behavior

Infill is leaving holes on small corners and around part holes that are visible on printed part.
Playing around with infill diameter does not help.
Slic3r Prusa looks  better using the same config although infill angles are rotated 90º.

[image]

[image]

[JC04PLAH025W05 Small.zip](https://github.com/alexrj/Slic3r/files/856024/JC04PLAH025W05.Small.zip)

lordofhyphens, 2017-03-20:
The prusa3d fork's rectilinear isn't the same as upstream rectilinear. Set to rectilinear2 on prusa3d and you'll be closer to comparing.

lordofhyphens, 2017-03-20:
Also need the model used, not just the config for the test case.

lordofhyphens, 2017-03-20:
You can play with fill_angle and see if that helps anything as well (it is 45 in your config)

alexrj, 2017-03-20:
I see that you have Infill Overlap set to 5%. The default in Slic3r is 55%. Try increasing it and let us know.

cerdeira, 2017-03-21:
Yes, 55% Overlap improves gap fill.
Have tried several options but so far did not get a good one.
But again, my experience is limited.

The stl file is this one.
[1 Sup XX ACME.zip](https://github.com/alexrj/Slic3r/files/858824/1.Sup.XX.ACME.zip)