slic3r/Slic3r #3782: Unconvenient behaviour with the Printer Connection and the Controller tab

msrd0, 2017-03-20:
### Version

Slic3r says: `1.3.0-dev`
git describe says: `1.2.9-772-ga69e4e39`

### Operating system type + version
ArchLinux rolling release (perl 5.24)

### Behavior
I have an Anycubic Prusa I3 printer running with custom Marlin 1.1.0-RC8 firmware. For compatibility with low-speed devices I reduced the baudrate to 57600. Today I downloaded the development version of Slic3r (I was using the 1.2.9 release before) and noticed there is only an option for 115200 and 250000 baud rate. Since switching to a higher baud rate is not an option for me I searched the source code and found the baud rate beeing added to two enums, one in [PrintConfig.cpp](https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.cpp#L1072) and one in [PrinterPanel.pm](https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GUI/Controller/PrinterPanel.pm#L131). I added my baud rate 57600 to both enums and compiled slic3r again.

I added `/dev/ttyUSB0` as the serial port and 57600 as the baud rate in the printer settings and clicked the test button. Slic3r reported that the connection is successfull. Moving to the Controller tab, I can confirm that the connection is established and that Slic3r is able to read the messages from the printer. Unfortunately, Slic3r is not able to send any messages to the printer (Cura and Pronterface both can). I can't tell whether this is because a bug in Slic3r or me messing up the source code by adding the baud rate.

Another issue that doesn't seem related to the baud rate is that I simply can't scroll down in the Console after clicking the Manual control button. As soon as I start moving the slider down, it starts flickering to the top all the time. I'd expect the Console to stay where I move the slider or at least automatically scroll down to the most recent message, but not to the top.

### `git diff`

```
diff --git a/lib/Slic3r/GUI/Controller/PrinterPanel.pm b/lib/Slic3r/GUI/Controller/PrinterPanel.pm
index 53850a69..214be444 100644
--- a/lib/Slic3r/GUI/Controller/PrinterPanel.pm
+++ b/lib/Slic3r/GUI/Controller/PrinterPanel.pm
@@ -128,7 +128,7 @@ sub new {
         my $serial_speed_sizer = Wx::BoxSizer->new(wxHORIZONTAL);
         {
             $self->{serial_speed_combobox} = Wx::ComboBox->new($self, -1, $config->serial_speed, wxDefaultPosition, wxDefaultSize,
-                ["115200", "250000"]);
+                ["57600", "115200", "250000"]);
             $self->{serial_speed_combobox}->SetFont($Slic3r::GUI::small_font);
             $serial_speed_sizer->Add($self->{serial_speed_combobox}, 0, wxALIGN_CENTER_VERTICAL, 0);
         }
diff --git a/xs/src/libslic3r/PrintConfig.cpp b/xs/src/libslic3r/PrintConfig.cpp
index c1b5870f..2ab6ad89 100644
--- a/xs/src/libslic3r/PrintConfig.cpp
+++ b/xs/src/libslic3r/PrintConfig.cpp
@@ -1069,6 +1069,7 @@ PrintConfigDef::PrintConfigDef()
     def->cli = "serial-speed=i";
     def->min = 1;
     def->max = 300000;
+    def->enum_values.push_back("57600");
     def->enum_values.push_back("115200");
     def->enum_values.push_back("250000");
     def->default_value = new ConfigOptionInt(250000);
```

lordofhyphens, 2017-03-20:
If it's an enumeration you shouldn't have to add it as a default, the combobox normally lets you write whatever you want in there too. 
I think that yes, having 57600 (also a legal and useful baudrate for more 3d printing applications) in the enum list is a good thing though.

msrd0, 2017-03-20:
Ok good thing I can edit baud rates, though it doesn't help me with the issue. I'm used to programs like minicom that replace unknown baudrates with 115200 without even telling me.

alexrj, 2017-03-20:
Serial communication has known issues on Linux. I'm trying to figure out what's wrong.

alexrj, 2017-03-25:
I added 57600 and fixed the serial communication on Linux.

closed 2017-03-25 by alexrj