slic3r/Slic3r #3783: option to select the Z slider unit, to mm or layer number

efa, 2017-03-21:
### Version
1.2.9

### Operating system type + version
Ubuntu 16.04 64 bit
Win7 64 bit

### Behavior
The Preview and Layers Z axis is in mm only.
Please add an option to select the Z slider unit, to mm or layer number

_Is this a new feature request?_
yes RFE

#### STL/Config (.ZIP) where problem occurs
all

xoan, 2017-03-21:
I vote for something like to put this information overlayed in the view, at the bottom right corner, probably by moving the current Z mm too.

alexrj, 2017-03-21:
The current unit is layers. Each step of the slider *is* a discrete layer, not an arbitrary Z in mm. The label below the slider shows the current layer's Z.
Layer numbers are not very meaningful, because support material can have interlaced layers and objects in the same print can use different layer heights.

Rather than just talking about the implementation, can you please focus on the test case and discuss *why* are you actually asking for such change? It's very unlikely that we do something if you don't share why you are proposing it.

lordofhyphens, 2017-03-21:
As I said in #3022, this information (the current layer number) is less useful than you would think once the assumption that all extrusions on the same Z belong to the same layer is invalid. 

That assumption is invalid. 

So you will need to explain more comprehensively what you want to do and why you want to do it. Preferably with pictures.

Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

maxdarkdog, 2018-04-17:
I do need the layer number for my colorchange script that uses layer number.... i cant understant why this kind of feature used in all other softwares is missing in here .... Really ... 
You may be find this useless but it is a must have feature for me...