slic3r/Slic3r #3784: Vase gcode inconsistent

jmgiacalone, 2017-03-21:
### Version
61c8359995e785a638d2f39390c2998a889b4a93

### Operating system type + version
Ubuntu 16.10 64bit

### Behavior
Solid object sliced in vase mode. Preview and Layers tabs display toolpath as expected. Gcode output produces GCode with errors. Screenshot shows the same object layer in Slic3r Layers tab and Pronterface GCode view:

[image]

This behaviour occurs with various models, all reported as having no mesh errors.
[golvend.zip](https://github.com/alexrj/Slic3r/files/858120/golvend.zip)

xoan, 2017-03-21:
It is due the use of relative E distances.

Did you try to print it and see if the results are as expected in the Slic3r preview?

jmgiacalone, 2017-03-21:
Brilliant, thank you. I had forgotten it could be related to that bug. I didn't print it as I saw no point given the errors in the Pronterface preview.
I have re-sliced the file without relative extrusion moves and all look s good.

closed 2017-03-21 by jmgiacalone

alexrj, 2017-03-21:
Hi @jmgiacalone, nice to hear from you!
I just pushed a fix for that bug.

jmgiacalone, 2017-03-21:
Hello @alexrj 
Thanks for the fix. I have pulled/compiled/sliced and it's printing now. All looks good!