slic3r/Slic3r #3785: 3D view confused about Z axis placement of objects after loading from AMF

kefir-, 2017-03-21:
### Version
1.3.0-dev 61c8359995e785a638d2f39390c2998a889b4a93

### Operating system type + version
Ubuntu 16.04

### Behavior
I added two objects to the plater, and for one of these objects I added a modifier and some part settings to apply to the modifier. Next I saved the plate as an AMF, to store the various settings.

After restarting slic3r, I loaded the AMF, and first I noticed that the objects are not dropped on the Z=0 plane, but they are dropped halfway down beneath. It didn't look like this before I restarted slic3r. Fortunately, prints seem to work fine anyway, printing from Z=0, so it's a cosmetic issue. Here's a screenshot:

[image]

Next, I wanted to replace the main part that had a modifier with a new version. I clicked on the object, added the new part, and then deleted the old part. My aim with this was to test whether I can replace parts of a complex plate, without having to go through all the setup from scratch again. Now the new part is dropped on Z=0, while the other part is still dropped halfway beneath. Again it seems to be cosmetic only, the preview and layers views show both starting from Z=0, so it's just an issue with the 3D view. Here's another screenshot:

[image]

Fortunately, the workflow of setting up a complex build plate with modifiers, saving to AMF, and then later loading the AMF and modifying objects seems to work fine!

lordofhyphens, 2017-03-21:
As a FYI I pushed a change a couple days ago that lets you export a single
object with its modifier meshes as an AMF.

kefir-, 2017-03-21:
@lordofhyphens that's interesting, and certainly useful if I can also add a single object like that to a plate. But this time just replacing the main part with a new stl was the workflow I was looking for. And it worked fine, except for the confusing Z offsets. :)

alexrj, 2017-03-23:
Should be fixed now. Thank you!

closed 2017-03-23 by alexrj