slic3r/Slic3r #3786: "Right-click -> Reload from disk" reloads wrong object and adds it instead of replacing it

kefir-, 2017-03-21:
### Version
1.3.0-dev 61c8359995e785a638d2f39390c2998a889b4a93

### Operating system type + version
Ubuntu 16.04

### Behavior
I have loaded an AMF plate, as described in #3785. It has two objects, one of which has a modifier and some separate parameters.

If I right-click one object and choose "Reload from Disk", I was a bit curious about what would happen. The AMF contains the original STL file name, but would that be reloaded or the entire AMF or just this object from the AMF? It's not really obvious given the situation.

Anyway, I'm not really sure what happens but Slic3r appears to add the other object once more, and place it on top of the selected object. Pictures will describe this better. Here's a screenshot before reloading:

[image]

and below is what it looks like after reloading, when I right click on the smaller part on the left to reload. If I right click the part on the right, the smaller part is overlaid on top of that instead, so it's always the other part that is placed on top of the selected part.

[image]

### Test files

Regarding test files, I'm not sure if I need all STLs in addition to the AMF, or how this process really works, so I'm not uploading anything right now.

alexrj, 2017-03-21:
Good catch.

alexrj, 2017-03-21:
Fixed. Thank you!

closed 2017-03-21 by alexrj

kefir-, 2017-03-21:
Thanks! But perhaps there should be some clarification of what is reloaded, I'm still not sure. :-) Is the original STL file reloaded?