slic3r/Slic3r #3789: New "send to printer" code seems broken

gege2b, 2017-03-22:
Hi all, 

### Version
Current master (bintray build 4c14c65  ) 

### Operating system type + version
windows 10 

### Behavior
* _Describe the problem_

When clicking "send to printer" button, the dialog is shown but nothing is done after clicking on OK

AND

If using "Alt+ Clic" on send to printer button, it works ONLY if "overwrite existing file" has been checked earlier. Otherwise, it check for existing file on the server and then throw a "401: Unauthorized" error
[image]

* _Steps needed to reproduce the problem_

- load a model
- click on send to printer
- change settings if needed
- click OK
Nothing is done, no file is uploaded.

or

- Load a model
- Click on ALT + Send to printer button
- If "Overwrite existing file" was checked, it'll work
- If "Overwrite existing file" was NOT checked, it'll throw the error above

* _Expected Results_

Uploading the Gcode with whatever method

* _Actual Results_

Upload of Gcode is done under certain circumstences 

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

model and config independant (except for octoprint settings, of course)

lordofhyphens, 2017-03-22:
Maybe permissions on the Octoprint side? If you're willing to sit down and spend some time with curl, a copy of the Octoprint API docs, and how the uploads are currently being done (look for usage of LWP::UserAgent) you could probably work out if it's something permissions related.

gege2b, 2017-03-22:
well, I didn't change anything on octoprint side and the problem appeared as soon as I downloaded the nightly

Upload works fine, as long as I check the overwrite option

Note : on Linux, the dialog works as expected, but I still get the same error with overwrite option disabled

As soon as I get some time at work I will dig more deeply with curl & co as suggested

alexrj, 2017-03-23:
I might have broken something, will check...

alexrj, 2017-03-23:
I pushed an untested fix... let me know!

gege2b, 2017-03-23:
Great ! Thank you !

I just tested under linux and it shows no error and works as expected. 

Will test tomorrow with the automated Win build, but I guess it would be OK too :)

Just FIY : Alt + Clic on KDE (with default config) is already used as a global shortcut to move the current windows. So the Alt+clic on "send to printer" don't work on KDE, until the default key modifier is changed from ALT to Meta (the "windows" key) in the windows management settings

lordofhyphens, 2017-03-23:
Yeah, we can't use win key on Windows, because it is reserved.

gege2b, 2017-03-24:
@alexrj Well, it almost work under windows with the current Windows nightly.

- If I use the ALT-CLIC method, it works fine without error, with whatever option enabled

- If I use the dialog, it just don't do anything (but show no error), it doesn't even fire the "export" of gcode

@lordofhyphens I was talking about the win key under KDE of course, I was not suggesting to use the Win key for this shortcut.
Maybe using SHIFT instead of ALT ? But I don't know if it will conflit with another OS :)

alexrj, 2017-03-26:
Okay, I switched to Shift instead of Alt.

I'll need to try reproducing the issue on Windows.

alexrj, 2017-04-04:
I fixed this. Thank you!

closed 2017-04-04 by alexrj