slic3r/Slic3r #3791: Excesive supports and background processing always enabled, failed tests

Tinchus2009, 2017-03-23:
### Version
latest built 1.2.9-780-g4c14c65

### Operating system type + version
win 10 64b

### Behavior

Im catching some issues with this built:
1.- the background processing is enable all the time, despite if it is not in the preferences options. Also, the the CANCEL button has disapeared (the one you could use to cancel the slicing)
2.- the support generation is super excesive. I remember I read something about support generation in the git log, may be this is related? I have atached some pictures and config files. The model used as example has already been printed, and it is ok, was printed with version 1.2.9-694-g306264c and that version didnt genereted that much of supports (images atached)
3- in the last part of build, the tests parts, I get some failed tests:
t\15_config.t ................... Possible attempt to separate words with commas at t\15_config.t line 289.
t\fill.t ....................... 90/93
#   Failed test 'infill is only generated under the forced solid shells'
#   at t\fill.t line 300.
t\fill.t ....................... 93/93 # Looks like you failed 1 test of 93.
t\fill.t ....................... Dubious, test returned 1 (wstat 256, 0x100)
Failed 1/93 subtests
Test Summary Report
-------------------
t\fill.t                     (Wstat: 256 Tests: 93 Failed: 1)
  Failed test:  90
  Non-zero exit status: 1
Files=34, Tests=529, 91 wallclock secs ( 0.17 usr +  0.16 sys =  0.33 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!

[falla.zip](https://github.com/alexrj/Slic3r/files/864459/falla.zip)

lordofhyphens, 2017-03-23:
Change support threshold from 0 to either 60% or 200%.

lordofhyphens, 2017-03-23:
One thing per issue, please.

Tinchus2009, 2017-03-23:
ok, i will edit and create separated issues, sorry

Tinchus2009, 2017-03-23:
60% produces the same rsult as the picture posted: an incredible amount of supports not necesary... if you increase that to 80% you get less support, yes but still unnecesary. supports are being built in overhang areas, but those areas get no benefit from supports. a lot of material and printing time is added with this new logic

lordofhyphens, 2017-03-23:
FYI If you want the old behavior, use 200%.

lordofhyphens, 2017-03-23:
The values are adjusting what used to be a constant in the auto generated supports  (when 0 was used). 2.0 (or 200%) was the original value of the constant and represents the % of perimeter width left with no extrusion directly below it.

Tinchus2009, 2017-03-23:
Yes, definitly I will use the old behaviour. Im not aware of the reasons for changeing to this new behaviour, but in my opinion is s step back, I cant see any any advantage of haveing those supports generated. The support is separated 0.2 mm from those perimeters in the overhangs... no perimeter should fall, and if it does that emans the printer is not working well. The problem with overhangs is the curling of edges, not perimeters droping (at least as I said if the printer is well calibrated). and if you print the model I psoted, the result is exactly the same with supports and without supports, meaning those supports are not usefull at all. And the amount is huge, like 30% of extra material and a lot more printing time....

lordofhyphens, 2017-03-23:
Came about as a response to someone else's PR/issue, as all things. They
wanted to change the constant to 0.5 (aka 50%). I might be amenable to
changing the default back to something like 150% or all the way back to
200%, but different printers are going to have different sweet spots.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Tinchus2009, 2017-03-23:
Well... I think the defsult should be 200% if that is the old behaviuor.
Still dont understand why this change. If youbtake a look, a value of 50%
is creating supports in overhangs of 45 degrees and less, wich should be
print by any printer without supports. Even 60% is creating supports to 45
fegrees overhangs... That is definitly wrong.

lordofhyphens, 2017-03-29:
If someone wants to experiment around a bit with a few different printers and work out a reasonable default that is between the old value (200%) and the current default (60%) be my guest.

Tinchus2009, 2017-03-31:
I have been doing some tests. I have generated some tests:
this is an overhang test, sine i have been observing the generation of supports where, in my opinion, it shouldnt be supports. the overhang test goes from a 40 to 65 degrees. my printer is a prusa i3.
Something is not working with the new support code i think:
1. if i set a 60 degrees threshold, support is generated for all the object, the 40 degrees overhang is supported, and definitly it shouldnt.
2. setting a value of 60% generates support starting from the 50 degrees overhang (we can debate if this is necesary, not for my printer). But the problem is the the interface layer are printed in the air, there is a huge gap between them and the support structure, in the real print this generates horrible bloobs as the best scenario, in other cases the interface layer just didnt printed in the correct place  (I printed 3 times the same object with the same config to have consistent conclusions)
3. The default OLD value, now AKA 200% generates no support at all, in my particular case, my rinter can print the max overhang value og 65 degrees really well, it only produces a very small detail in 1 of the corners (this is more a thing of acceleration than an issue related to supports, because the detail it is in only 1 of the corners, the other corner of the 65 degree overhang is perfect).
4- a Vaalue os 120% produces supports only in the 65 degree overhang, for me, this would be my default may be, 
I have uploaded pictures of everything.
Some conclusions: useing degrees values in the threshold value seems to not work, been conservative, if you set a value of 45 degrees (where in theory you dont need supports) created supports even for overhangs of 40 degrees.. Also, the default value of 60% seems ok since it starts generating support starting at the 50% overhang, for some printer this could be ok, in my case it is totally unnecesary. But there is a problem here  and it is that the interface layers have a GAP between them and the support structure, this make them somehow fail. 
[falla.zip](https://github.com/alexrj/Slic3r/files/884194/falla.zip)

lordofhyphens, 2017-03-31:
Thank you for your efforts.