slic3r/Slic3r #3792: Keyboard shortcuts doesn't work in Preview/Layer tab

xoan, 2017-03-23:
### Version
current master

### Operating system type + version
Arch Linux 4.9.16-1-lts x86_64
Xfce 4.12

### Behavior
Keyboard shortcuts doesn't work in `Preview`/`Layer` tab

### Steps needed to reproduce the problem
1. Open Slic3r
2. Change to `Preview` or `Layer` tab
3. Try `Ctrl + O` (_Open STL/OBJ/AMF..._) or another keyboar shortcut you want

### Expected Results
Action associated to the keyboard shortcut should be triggered.

### Actual Results
No action.

### Workaround
If the focus is in some widget (a dropdown box, a tab...), keyboard shortcut works. It can be achieved by clicking twice in `Preview`/`Layer` tab, and then, `Ctrl + O` triggers the action.

[image]

If you change the focus by clicking the `Print settings` or another dropdown box in the `Plater`, then change to `3D`/`2D` tab, and then back to `Preview`/`Layer` tab, issue is addressed again. 

This workaround is not needed when in `3D`/`2D` tab.

alexrj, 2017-03-23:
I can't reproduce this on OS X, it might be Linux-specific...

lordofhyphens, 2017-03-24:
So you're sure that the Slic3r window had focus then? 

IIRC doubleclicking on preview forces another slice.

Lenbok, 2017-03-31:
Still present on master 8f789da9 post gui3-merge on Ubuntu 16.04.

Clicking on a top tab (e.g. Plater/Controller) usually has a visual indication of selection status (as shown in the animated image above), but if you click on the lower tabs (3D/2D/Preview/Layers) you only get the visual indication of selection status if that tab is already the one that is frontmost. Otherwise you don't get the visual indication, and for the Preview/Layers tabs the keyboard shortcuts are broken in this state.

xoan, 2017-05-08:
I think I found the reason for this issue.

Both 3D Preview and Layers tab have an EVT_KEY_DOWN event handler that looks for a keycode, but doesn't skip the event processing when another key is pressed.

Documentation: http://docs.wxwidgets.org/3.0/classwx_event.html#a98eb20b76106f9a933c2eb3ee119f66c

```diff
diff --git a/lib/Slic3r/GUI/Plater/2DToolpaths.pm b/lib/Slic3r/GUI/Plater/2DToolpaths.pm
index c7b3765c..2c5a4a7d 100644
--- a/lib/Slic3r/GUI/Plater/2DToolpaths.pm
+++ b/lib/Slic3r/GUI/Plater/2DToolpaths.pm
@@ -60,6 +60,8 @@ sub new {
         } elsif ($key == 68 || $key == 317) {
             $slider->SetValue($slider->GetValue - 1);
             $self->set_z($self->{layers_z}[$slider->GetValue]);
+        } else {
+            $event->Skip;
         }
     });
     
diff --git a/lib/Slic3r/GUI/Plater/3DPreview.pm b/lib/Slic3r/GUI/Plater/3DPreview.pm
index dddefee0..8fa9188e 100644
--- a/lib/Slic3r/GUI/Plater/3DPreview.pm
+++ b/lib/Slic3r/GUI/Plater/3DPreview.pm
@@ -58,6 +58,8 @@ sub new {
         } elsif ($key == 68 || $key == 317) {
             $slider->SetValue($slider->GetValue - 1);
             $self->set_z($self->{layers_z}[$slider->GetValue]);
+        } else {
+            $event->Skip;
         }
     });
```

I've tried in GNU/Linux with wxGTK 3.0.3 and it works. I don't know if it has some side effects in another platforms, but it shouldn't.

PR #3941 (2017-05-10): Skip event handling in 3D/Layer preview tab when no UP or DOWN key is pressed
Fixes #3792 by skipping event handling in  3D Preview and Layer tabs when no `UP` or `DOWN` key is pressend.
closed 2017-05-10 by alexrj