slic3r/Slic3r #3793: background procesing always enabled

Tinchus2009, 2017-03-23:
### Version
latest built 1.2.9-780-g4c14c65
Operating system type + version

win 10 64b

the background processing is enabled all the time, despite if it is not in the preferences options. Also, the the CANCEL button has disapeared (the one you could use to cancel the slicing)

alexrj, 2017-03-23:
I'm parking this as we're going to merge the gui3 branch soon which contains many changes which should also fix this.

alexrj, 2017-03-26:
@Tinchus2009, I can't reproduce this. Can you `git pull` and try again?

Tinchus2009, 2017-03-27:
git pull executed... now it is strage... all tabs have desapeared, the only visible tab is the plater tab. Accesing printing/ filament and printer tab its posible through the settings menu.
Also, if I change any value ofr example in the printing settings, the background processing is executed, despite if it is enable or disable in the preferences.

Build version: 1.2.9-838-gc347487

You want me to try building from scratch?

lordofhyphens, 2017-03-27:
The  gui3 branch was just merged in.

I did notice that it triggers processing on close of the profile dialog.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2017-03-29:
I can't reproduce this on Linux. If I change options with background processing off, I get moved back to the 3d tab when the dialog is closed (as expected), slicing doesn't begin until I click the preview tab.

lordofhyphens, 2017-03-29:
@Tinchus2009 
This is the new UI (so it's normal for now). 
[image]

If I change options with background processing on, it acts as expected (remaining on the preview tab and immediately reslicing).

Tinchus2009, 2017-03-29:
It has been fixed now with the latest build. I close the report then.

closed 2017-03-29 by Tinchus2009