slic3r/Slic3r #3794: failed tests

Tinchus2009, 2017-03-23:
### Version
latest built 1.2.9-780-g4c14c65

 in the last part of build, the tests parts, I get some failed tests:
t\15_config.t ................... Possible attempt to separate words with commas at t\15_config.t line 289.
t\fill.t ....................... 90/93
Failed test 'infill is only generated under the forced solid shells'
at t\fill.t line 300.
t\fill.t ....................... 93/93 # Looks like you failed 1 test of 93.
t\fill.t ....................... Dubious, test returned 1 (wstat 256, 0x100)
Failed 1/93 subtests
Test Summary Report

t\fill.t (Wstat: 256 Tests: 93 Failed: 1)
Failed test: 90
Non-zero exit status: 1
Files=34, Tests=529, 91 wallclock secs ( 0.17 usr + 0.16 sys = 0.33 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!

alexrj, 2017-03-23:
Interesting, that test is not failing on any of our CI server, not I can reproduce it...
What's your setup? Operating system, Perl version any any other detail.

Tinchus2009, 2017-03-23:
Sorry, missed that data:

win10 64b
strawberry perl: perl 5, version 18, subversion 2 (v5.18.2) built for MSWin32-x64-multi-thread (the one recomended in the wiki for slic3r)

alexrj, 2017-03-23:
Thank you. I noticed it happened randomly a couple times in our Windows build server as well.

teemuatlut, 2017-03-23:
Can confirm, also on Win10 x64 with Strawberry Perl v5.24.0 x32.

NateTG, 2017-03-23:
I got a similar test failure.
#   Failed test 'infill is only generated under the forced solid shells'
#   at t/fill.t line 317.

Debian 9.0, Perl 5.20.2

Tinchus2009, 2017-03-24:
I forgot to mention that despite of the failed tests, the build completed succesfully and slic3r started with no problems

corthiclem, 2017-03-30:
Hi everyone, 

I have the same issue for few days now on Windows 7 with Perl 5.18
The failed test seems to be related to the setting "Solid Infill threshold area" and, in fact, even though the setting is set to 1000 mm^2, solid infill is **not computed** on a 15mm square cube. 

What is odd is that when I tried to compile Slic3r on Linux, the test passed and the solid shells are computed correctly. 

Has someone any idea on the cause of this error?

Thanks in advance

lordofhyphens, 2017-03-30:
@corthiclem

I haven't had a chance to look into it very closely. 

Btw, this will be relevant to your interests:
https://github.com/alexrj/Slic3r/issues/3815

alexrj, 2017-04-04:
I pushed a fix for this. @Tinchus2009, @corthiclem, can you confirm it works now?

Tinchus2009, 2017-04-04:
it is ok (build version  1.2.9-914-g557f1e2) :

All tests successful.
Files=34, Tests=533, 81 wallclock secs ( 0.05 usr +  0.16 sys =  0.20 CPU)
Result: PASS

closed 2017-04-04 by alexrj

corthiclem, 2017-04-05:
@alexrj the tests passed and the feature "solid infill threshold area" works correctly :) 

Thanks for this fix!