slic3r/Slic3r #3798: Scaling is applied inconsistently over copies

NateTG, 2017-03-24:
### Version
Version 1.3.0-dev

1.2.9-790-gaa02dfe2

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* Multiple stls, copy, scale, and rotate interact in some way that causes inconsistent scaling.  I haven't gotten a minimal case,  it seems to happen when I have copies of 2 different STLs on the plate, 2 of one and 3 of the other and then do rotating, scaling, and copying in some random order.

* I expect all copies of an obeject to have the same scaling.
* Sometimes they don't:

[image]

[config.zip](https://github.com/alexrj/Slic3r/files/866882/config.zip)

alexrj, 2017-03-26:
I am not able to reproduce this.

NateTG, 2017-03-27:
OK:
Go to 2D plating view
Add Object A.
Hit "more" twice.
Rearrange to remove overlap
Scale by 200%
Add Object B
Scale by 200%
Hit more once.
There should be a random arrangement of object A and object B on the plate now.
Select object A
Right click -> rotate -> around X axis -> 180
Switch to the 3D view.  You should see something odd.

Tinchus2009, 2017-03-29:
I culd reproduce this following your steps. At the end , when you switch to the 3d platter, the object A is OK, but its 2 copies have been scaled like 400% and the original A only to teh expected 200%. In the 3d platter if you scale object A once more, its 2 copies will be adjusted to the correct size again.

NateTG, 2017-03-29:
It looks like
scale (not 100%) -> rotate (180) -> scale 
will pop up a menu with 100%.  That's probably related.

alexrj, 2017-04-04:
Thank you for the instructions! I was able to reproduce this and fix it.

closed 2017-04-04 by alexrj