slic3r/Slic3r #3799: [Feature Request] Extended Context Menu and Toolbar: Rotate Object in X or Y

foreachthing, 2017-03-24:
### Version 1.3.0-dev-5d2626f 

### OS
Windows 7 Pro

### Current behavior
* Import STL
* STL is wrong -> Z is not up (this does happen, you know!)
* Use context menu
* Choose Rotate ...
* Around X (or Y) ...
* Enter Value [ENTER]

### New behavior
* Import STL
* STL is wrong -> Z is not up
* Click button in toolbar (only in X so far) or use context menu to quickly rotate object in X or Y by 90° or 180°.

_Is this a new feature request?_
Yes.

_Plater.pm:_
I made such changes to the `\lib\Slic3r\GUI\Plater.pm` - I'm just not familiar with github to make a proper pull-request.
Please review this file attached and let me know what you think.
[ExtendedContext.zip](https://github.com/alexrj/Slic3r/files/867308/ExtendedContext.zip)

lordofhyphens, 2017-03-24:
1.Fork slic3r on github.
2. Using either the github client or the regular git client, checkout your
fork and make a new branch.
3. Make your changes
4. Commit changes to your branch.
5. Push branch to your github fork.
6. On the upstream github, click "create pull request" button.
7. For the head, select your fork and then the feature branch from the drop
down.

https://danielmiessler.com/study/git/#gs.3WbC74c

foreachthing, 2017-03-24:
@lordofhyphens: Thanks for the "walk-through"!

lordofhyphens, 2017-03-24:
@foreachthing no problem. I am going to close this issue and you can look at the comments from myself and @alexrj on the pull request. 

If you need to update the PR to use a new master, you should use git rebase.

short answer:
Setup
````
git remote add upstream https://github.com/alexrj/Slic3r
git fetch upstream
git checkout -b upstream/master upstream-master
````
On every update:
````
git checkout upstream-master 
git pull
git checkout <branchname>
git rebase upstream-master
````
This will apply your commits to the upstream master branch. 

Reference:
https://git-scm.com/docs/git-checkout
https://git-scm.com/docs/git-rebase

closed 2017-03-24 by lordofhyphens