slic3r/Slic3r #3800: [gui3] Dirty overrides removed when close Print Settings dialog

xoan, 2017-03-24:
### Version
[gui3 branch](/alexrj/Slic3r/tree/gui3) 70395257

### Operating system type + version
Arch Linux 4.9.17-1-lts x86_64
Xfce 4.12

### Behavior
When some of dirty overrides was added, open the _Print Settings_ dialog and then closing it makes disappear the prior added dirty overrides.

### Steps needed to reproduce the problem
1. Open Slic3r
2. Add some overrides using the `+` buton in the right pane
3. Open Print Settings by clicking in the `gear` button near `Print Settings` combo or by `Ctrl + 1`
4. Close this dialog

**Note.** These are the most simple steps, not involving the fact that you can made some changes in the _Print Settings_ that implies a change in the dirty overrides values.

### Expected Results
Dirty overrides should remains in the right pane.

### Actual Results
Dirty overrides are removed from the right pane.

gege2b, 2017-03-24:
I can reproduce it on Kubuntu as well

lordofhyphens, 2017-03-24:
Can you add this issue id to the pull request thread?

msrd0, 2017-03-25:
I can reproduce this as well (ArchLinux / i3wm). This also happens when restarting Slic3r.

Both issues only happen when I added the overrides with the green plus icon on the right side of the platter. When I add the overrides in the Print Settings, they remain, no matter whether I open the Print Settings again or restart Slic3r.

alexrj, 2017-03-25:
Good catch! I fixed this.

alexrj, 2017-03-25:
@msrd0, those overrides added on the fly are discarded when restarting Slic3r or selecting another preset. This is intentional.

closed 2017-03-25 by alexrj