slic3r/Slic3r #3801: [Feature Request] Configurable material for support interface

RafaelEstevamReis, 2017-03-24:
### Version
None

### Operating system type + version

### Behavior
Some support materials are very expensive ou hard to print.
Would be nice to have an option to print the support with default (cheap - abs?) material and only the interface of support-model with expensive (soluble?) material.
This approach do not work well with internal supports but for big models with simple overhang will decrease the cost of print.

Slic3r already has an option for support material, and additional option for interface material would be sufficient in configuration.
In code slic3r already differentiate support body from it's interface.

_Is this a new feature request?_
Yes

#### STL/Config (.ZIP) where problem occurs
Any model with a lot of support outside of model

RafaelEstevamReis, 2017-03-24:
Wait... I think that already has it ...
Multiple Extruders -> Suport/Raft interface extruder

alexrj, 2017-03-24:
Exactly, it's there. Enjoy!

closed 2017-03-24 by alexrj

lordofhyphens, 2017-03-24:
Serves me right for looking at stuff after I just got up.

RafaelEstevamReis, 2017-03-24:
Sory for this, I looked at support section and I did not even notice that
were on Multi Extruder tab