slic3r/Slic3r #3803: Slic3r not opening when executing stl file

diegoolavarria, 2017-03-24:
### Version
Prusa Version

### Operating system type + version
Windows 8.1

### Behavior
When I asociated .stl files into executing Slic3r, is expected to open Slicer with the .stl file already loaded, but it creates a .gcode output file without opening the software. Is it a common problem?

xoan, 2017-03-24:
This issue tracker is for upstream Slic3r version. You must report it in the Prusa's fork: https://github.com/prusa3d

lordofhyphens, 2017-03-24:
Closing. If you can reproduce with current build from http://dl.slic3r.org/dev I will reopen.

This is, however, probably normal behavior. If we don't do that then programs like repetierhost will break (and all of the command line interface).

lordofhyphens, 2017-03-24:
The prusa3d fork uses 2 shell execs, one with a command line window and one without. While we could adjust shell.cpp to build two copies one that forces --gui and one that doesn't (slic3r-gui.exe and slic3r.exe), I'm sure someone else would get confused. 

Reclosing until @diegoolavarria provides a test case with upstream slic3r.

diegoolavarria, 2017-03-27:
Ok I created an executable of the source code provided here, and tested by executing the .stl file into the slic3r.exe. It opened a command prompt window (which it happens when you execute a .exe created with perl), showing the process of creating a .gcode file.

[image]

The final result is a .gcode file with the presets.

alexrj, 2017-03-27:
How did you create the executable?

diegoolavarria, 2017-03-27:
pp -x -o slic3r.exe slic3r.pl

alexrj, 2017-03-27:
Then it's normal.
See the contents of the package/win directory for how to properly package Slic3r. Otherwise just get a fresh prebuilt one at http://dl.slic3r.org/dev

diegoolavarria, 2017-03-27:
Yes, I've tried with different versions and releases, so I also figured out it's a normal setting. Although, I'm trying to configure the code so when I double click a .stl file, it opens Slic3r showing the model instead of automatically create a .gcode file.

lordofhyphens, 2017-03-27:
@diegoolavarria I have a

diegoolavarria, 2017-03-27:
@lordofhyphens slic3r.exe by itself always opens the GUI. The problem is when executing a .stl file asociated with slic3r.exe

xoan, 2017-03-27:
Don't know if it has something to do, but in GNU/Linux, `--gui` option should be passed to `slic3r.pl` when a `file.stl` is provided in order to launch it instead of process it in the command line.

```zsh
% slic3r file.stl       # this process file.stl in the command line
% slic3r --gui file.stl # this opens slic3r and load file.stl
```

lordofhyphens, 2017-03-27:
I know, I haven't fixed that yet

lordofhyphens, 2017-03-27:
*sigh*

I will fix it. Please wait for me to say it is fixed before continuing on.
Hopefully that will be tonight.

PR #3804 (2017-03-28): force --gui for shell during compilation if -DFORCE_GUI is set
Fixes #3803
commit fe9d2946 (2017-03-28): Merge pull request #3804 from alexrj/lordofhyphens-split_gui

force --gui for shell during compilation if -DFORCE_GUI is set for compile wrapper. Fixes #3803
closed 2017-03-28 by lordofhyphens