slic3r/Slic3r #3806: Add a Notes Tab to the Printer Setting

ProfDrYoMan, 2017-03-26:
Hello,

It would be great to add notes to a printer setting.

lordofhyphens, 2017-03-26:
There's the start/end gcode field. Start lines with ; and you are good to
go.

ProfDrYoMan, 2017-03-26:
That's right. It's a "workaround". All other tabs got a "notes" section.

singh1114, 2017-03-26:
@ProfDrYoMan Why do we need notes for the printer settings?

ProfDrYoMan, 2017-03-26:
I have different setups vor min/max layer limits which I use for some projects.
Also two different g-code setup (one leaves the headbed on).
Yes, it is possible to "code" the differences into the profile name, but it get's to much for me.

lordofhyphens, 2017-03-26:
Fyi you can mark (as of today) print and filament profiles as compatible
with printer profiles.

lordofhyphens, 2017-03-26:
Not a replacement, but related.

lordofhyphens, 2017-03-29:
@ProfDrYoMan What's the intended use for this field? Do you just want it as reference for yourself in the GUI?

Should it get written to output Gcode as comments?

ProfDrYoMan, 2017-03-29:
My intend is to store information on what is special about this printer setup.
Same as for the notes field in the filament or print setup.

lordofhyphens, 2017-04-03:
Done.

commit c110a9d5 (2017-04-03): Adds printer notes item and tab. Fixes #3806
closed 2017-04-03 by lordofhyphens