slic3r/Slic3r #3807: Feature request: skirt for the second hotend

Remcatfoil, 2017-03-26:
### Version
1.2.9

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
win 10
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/
Thanks for this very good software.
I build a two hotends printer. When i print in bicolor, the skirt function only apply for T0 hotend.
It would be great to have a skirt function for the second hotend to prepare the filament of T1, before it  start printing. 
Best regards,
Remy

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

ignatz42, 2017-05-07:
Currently, in 1.3, the first layer of the skirt prints mixed between the extruders. I noticed that when one extruder isn't used until ~20 layer and then only sparingly, the result seems under-extruded for that extruder for some time. I also thought that priming both extruders in the skirt would be a good solution, so I made the following change to test:

`
C:\src\github\Slic3r>git diff lib\Slic3r\Print\GCode.pm
diff --git a/lib/Slic3r/Print/GCode.pm b/lib/Slic3r/Print/GCode.pm
index 710125c..6e6268d 100644
--- a/lib/Slic3r/Print/GCode.pm
+++ b/lib/Slic3r/Print/GCode.pm
@@ -471,8 +471,8 @@ sub process_layer {
                 # just use the 'skirts' setting; also just use the current extruder
                 last if ($layer->id > 0) && ($i >= $self->print->config->skirts);
                 my $extruder_id = $extruder_ids[($i/@extruder_ids) % @extruder_ids];
-                $gcode .= $self->_gcodegen->set_extruder($extruder_id)
-                    if $layer->id == 0;
+                $gcode .= $self->_gcodegen->set_extruder($extruder_id);

                 # adjust flow according to this layer's layer height
                 my $loop = $skirt_loops[$i]->clone;
`

The result was mixed. Quality for the second extruder increased. The gcode file increased in size by ~50% and print time was also much longer due to extra tool changes. I think these cons might be acceptable but the print quality of the skirt is better than quality for the object inside and I suspect this is due to decreased cooling. I'd like to try an external/separate priming tower next.

dobrzann, 2017-11-20:
i also think that skirt for second extruder can be very helpful in my case too.
e.g. You start to print small part without any raft, but you have model perimeters (extruder 0) and support structure (extruder 1) of the first layer. Extruder 0 starts from skirt and compensate retraction or previous material leaks but extruder 1 do not, and in the result support structure is printed form 50% or not at all (first layer of course).

Is it possible to add this function for extruder 1 too?