slic3r/Slic3r #3808: fully customizable z-lift issue/feature request

nicksears, 2017-03-26:
### Version
1.33.8 prusa edition, I know this is not the most relevant, but until the next release has gapfill fixed, there's no way I'm going to use it.

Win 10 32/64

### Behavior
I've turned on z-lift only above/below and it does not behave as I'd expect. 
The tooltip says this means the above value lets it skip the first above layer X
and the below value limits it to below Y layers

And this is fine if you need that, but I would think people would want the opposite.
I want hop ON for the first X layers to prevent peeling up a corner.
And then OFF until above a certain height when it's back ON to minimize deflecting of tall/thin parts

And it should just work with the appropriate values - when you make the "above" number higher than the "below" number, it should turn hop ON outside those values.

and file/stl, jst has to do with the hop only above/below

lordofhyphens, 2017-03-26:
Seriously, use the dev builds. https://dl.slic3r.org/dev

Pick the top entry in the version list.

There isn't anything special going forwards about our dev builds I'm how
they are packaged compared to what the release will look like. Nor do you
have to do anything with git.

lordofhyphens, 2017-03-26:
The hop avoidance if memory serves is to avoid hopping outside of max z.
On Mar 26, 2017 3:59 PM, "Joe Lenox" <lordofhyphens@gmail.com> wrote:

> Seriously, use the dev builds. https://dl.slic3r.org/dev
>
> Pick the top entry in the version list.
>
> There isn't anything special going forwards about our dev builds I'm how
> they are packaged compared to what the release will look like. Nor do you
> have to do anything with git.
>

lordofhyphens, 2017-03-26:
What it sounds like is that you want to be able to turn zhop on/off with modifier meshes.

nicksears, 2017-03-26:
Dev builds do not work with repetier (no exe). Also this has no effect on
the issue. I could (should) have just said I use 1.2.9 because it doesn't
matter.

I want to be able to turn on hop below AND above certain heights rather
than above BUT below.

lordofhyphens, 2017-03-26:
Yes they do, we provide shell exes just like the prusa3d fork.

nicksears, 2017-03-26:
K, thx I'll try again!

lordofhyphens, 2017-03-26:
I will see what breaks if I move zhop over to be enabled for modifier meshes.

lordofhyphens, 2017-03-27:
Short answer: everything. There's quite a bit of refactoring involved to make this work.

nicksears, 2017-03-27:
just to clarify because modifier meshes doesn't seem like the right term to describe what I'm thinking.
I want either a check box to hop inside or outside the above/below values (or it should honestly do it automatically if above is larger than below...
[image]

lordofhyphens, 2017-03-27:
I considered modifier meshes because there is the slab mesh and it would be a lot more flexible.