slic3r/Slic3r #3810: Changes to the Platter switches Tab

foreachthing, 2017-03-27:
### Version
Slic3r 1.3.0-dev
slic3r-master.2017.03.26.671.c347487

### Operating system type + version
Windows 7 Pro

### Behavior
* Activate **_Layers_**
* Make changes to the platter (i.e. enable/disable _Generate Support Material_)
* Tab **_3D_** gets selected.

Expected behavior:
* Do not change tab while making changes to the platter.

This is with every model and every setting I've found so far.

lordofhyphens, 2017-03-27:
This is intended, changing any settings value invalidates the preview (2d and 3d).

closed 2017-03-27 by lordofhyphens

foreachthing, 2017-03-27:
>   changing any settings value invalidates the preview

I agree.
Still, it could just update the preview as it does when the user switches the tab intentionally. The user will most likely change tab back to one of the preview tabs anyway.

alexrj, 2017-03-27:
If you want that behavior, just enable Background Processing in the application Preferences.