slic3r/Slic3r #3811: Unable to upload to OctoPrint

jammi, 2017-03-27:
### Version
slic3r-1.3.0-dev-426e79c

### Operating system type + version
macOS Sierra 10.12.3

### Behavior
* OctoPrint support no longer works
* Clicking on the Send to printer button and accepting the dialog, I get this error message: http://i.imgur.com/A7nYAyj.png
  * On subsequent tries (until restarting Slic3r) I get this: http://i.imgur.com/kilbzY8.png
* Nothing found on octoprint's log, so I'd guess Slic3r doesn't even try accessing it.
* Octoprint is at its newest release version (updated within a few days).

This issue is new to me. I used Prusa's builds for a while and the last stable official Slic3r before that. Neither had this issue.

[config.zip](https://github.com/alexrj/Slic3r/files/872308/config.zip)

alexrj, 2017-03-27:
Thank you for this report, it might be related to our new build procedure not packaging all the required modules for HTTP.

alexrj, 2017-04-04:
I fixed this. Thank you!

closed 2017-04-04 by alexrj

Astroman, 2017-04-14:
I just tried the latest development build (slic3r-1.3.0-dev-b78ccc6-.dmg) and have exactly the same problem. Should this be fixed in the dev build?

jeffbearer, 2017-04-19:
I too am seeing this in the slic3r-1.3.0-dev-075adca dmg

jammi, 2017-05-27:
This issue still (or again) persists.

rmpel, 2019-07-14:
I'm having similar issues in slic3r-1.3.1-dev-8adf792-osx
No error messages, but no uploads either. Release 1.3.0 works perfectly, so I know it's not my OctoPrints.