slic3r/Slic3r #3813: Pronterface compatibility broken because presets are saved without extension in slic3r.ini

xoan, 2017-03-27:
### Version
current master

### Operating system
Arch Linux 4.9.16-1-lts x86_64
Xfce 4.12

### Behavior
The way the default presets are saved now in `slic3r.ini` breaks the compatibility with the Slic3r integration in Pronterface:

* https://github.com/kliment/Printrun/blob/master/printrun/pronterface.py#L2171

Because Pronterface expects the preset name contains the full file name including `.ini`.

Probably it is an issue to be solved in Pronterface, right?

alexrj, 2017-03-27:
Oh, I didn't know that. I thought it only used the public CLI commands. I think it's easy enough to modify Pronterface to append `.ini` if it's not there...

lordofhyphens, 2017-03-27:
@kliment FYI...

alexrj, 2017-05-01:
I fixed this in https://github.com/kliment/Printrun/pull/817
I could merge it into the Printrun repository myself, but I can't test it at the moment. @xoan, if you run Pronterface from git would you have a change to test my change?

xoan, 2017-05-01:
Sorry, it doesn't work. It still updates `slic3r.ini` with full preset filename, including extension.

To solve this issue, it probaly involves some more changes in `load_slic3r_configs` and `read_slic3r_config`.

I can take a look int it this night. Let me try.

xoan, 2017-05-02:
OK, I just made a PR with changes that take into account the Slic3r version from `slic3r.ini`.

It is tested in GNU/Linux, with both `1.2.9` and `1.3.0-dev` Slic3r version.

alexrj, 2017-05-16:
Thank you for contributing that @xoan. I merged the PR in Printrun :)

closed 2017-05-16 by alexrj

xoan, 2017-05-16:
Nice :)

ntoff, 2017-07-09:
This seems to break integration with the Prusa edition of slic3r (1.35.5), with slic3r integration enabled, it appends .ini to the file, leading to being unable to locate the profile.

>Cannot find specified configuration file (C:\Users\redacted\AppData\Roaming\Slic3r\print\default.ini.ini).

bubnikv, 2017-07-10:
There is a similar issue posted on the Prusa3D fork:
https://github.com/prusa3d/Slic3r/issues/367

xoan, 2017-07-14:
It should be solved by patching Prusa edition of Slic3r to mimic the current Slic3r behavior.

ntoff, 2017-07-15:
or more easily, I just added a check to pronterface to not add .ini to the profile if it detects the prusa edition of slic3r