slic3r/Slic3r #3814: "Generate support" override is out of sync with its "print settings" bro'

gege2b, 2017-03-27:
Hi there

I'm not sure it's a bug, maybe it's an intentional behaviour, but it's a bit confusing

### Version
_Version of Slic3r used goes here_
acda91322cc651fb79eb6977d725d022e5c36b1b

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

Linux Mint Serena

### Behavior
* _Describe the problem_

Options "Generate support" aren't in sync.

* _Steps needed to reproduce the problem_

Load a model needing support to be printed
Go in print settings > support and check "generate support material", close the dialog
Switch to 3D preview or wait for background process to finish
No support present.

Note : this workflow is useful when you want to change other support settings for whitch you don't have overrides

* _Expected Results_
The "Generate support material" override to be synced with its clone in print settings, when closing the setting dialog,
the support to be generated

* _Actual Results_

The two mentionned settings are out of sync, and it seems that the support generation is only triggered by the state of the override 

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

model doesn't matter, default config

alexrj, 2017-03-28:
I'm not getting this. The override takes precedence over the profile option (hence the "override" name)...

lordofhyphens, 2017-03-28:
The problem is that it defaults to having the override active, which
confuses people.

I think the UX expectation here is that if you change it in the profile,
the override is updated to show current state. The user would then expect
to change the state from the override and have it last until the next
profile change.

gege2b, 2017-03-28:
@lordofhyphens yep, you get it

given the name, I agree that it could not be considered as a bug, but I guess that a lot of people won't find this behaviour ergonomic, because they will see these overrides more like "shortcuts" to settings

alexrj, 2017-03-29:
Okay, I got the point. I now pushed a change, and whenever the preset editor is closed the overrides are re-set to the preset values.

closed 2017-03-29 by alexrj

xoan, 2017-03-31:
Sorry to comment out of this when it is closed.

I think the correct behavior for this is to change **only** the parameters that have been changed in preset editor, and not to reset all of the overrides, IMO.

The way it is now implemented, if I change a lot of presets in the plater, and I open the Print Settings editor and then I close it, all this changes are lost.