slic3r/Slic3r #3815: LOOK HERE IF SLIC3R NO LONGER BUILDS ON 64BIT WINDOWS

lordofhyphens, 2017-03-28:
Slic3r will *NOT* build with basic Strawberry Perl 5.24 on 64-bit systems.

It will hang during the test execution phase. The reason for this is NOT a bug in Slic3r, but a combination of Perl exceptions and GCC versions > 4.8 and SJLJ exceptions. 

Unfortunately, the last known version of Strawberry Perl that works is 5.18. which ships GCC 4.7, which does not support all of C++11.

I've since built a stripped-down distribution of Strawberry Perl that instead uses GCC 6.3.0 and SEH exceptions. This will hold us until Strawberry uses SEH exceptions in its packaged version of GCC. 

* Install https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-perl-5.24.1.4-64bit.msi instead. You will need to uninstall Strawberry Perl prior to this.
    * Alternatively, overwrite C:\Strawberry with https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-perl-5.24.1.4-64bit.zip

* Download and extract https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-perl-dependencies-5.24.0-win-seh-gcc6.3.0-x64.7z into your Slic3r directory. 
    * You want to do this unless you really want to go through the pain I did preparing it (hint: it includes hacking wxperl to work with wxWidgets 3.1.0)

Any new issues that relate to the tests hanging forever will be closed and pointed here. I will then lock the conversation.

The wiki has been updated.

lordofhyphens, 2017-03-28:
@Tinchus2009

foreachthing, 2017-03-28:
Also slic3r-master.2017.03.28.686.fe9d294 from https://bintray.com/lordofhyphens/Slic3r/slic3r_dev will no longer run.
System Error: The program cannot be started due to missing libgcc_s_seh-1.dll .... reinstall ... [OK]

alexrj, 2017-03-28:
Perhaps we should detect Strawberry Perl >= 5.24 in Build.PL and display a warning there.

Tinchus2009, 2017-03-28:
THANKS for sending me the warning. I will Apply @lordofhyphens modifications to strawberry, and yes, I think what @alexrj proposes makes sense and will avoid people start posting the failure and comeing here quickly.

Tinchus2009, 2017-03-28:
WARNING to windows users:
Dont "git pull" for new dev updates. The new modified strawberry seems to breake something, slic3r didnt build in my system and worst: previous version of slicer dont start anymore. So basically: you will loose the ability to use any version of slic3r

Overwriting strawberry didnt worked. reinstalling strawberry with the provided MSI instalation file have some problems, the first one is that doesnt come with cpan, you have to instal cpanm again. Also slicer doesnt build either and any version of previously builted slic3r doesnt start anymore

lordofhyphens, 2017-03-28:
@Tinchus2009 I said to use the dependency archive (avoid building GUI from
cpanm, it will not work.  I have some limited instructions for building
wxperl with the new stuff but it is probably fragile).

Tinchus2009, 2017-03-28:
I did use that dependency package... But I never got to the step of building the GUI. The build fails at the first "perl build.pl" step
What Im missing? 
I followed these steps:
-unistalled strawberry
-installed your custom strawberry (I tried both: overwriting the old and the MSI file)
-downloaded and extracted the dependencies package to my test slic3r installation (latest buil was like 24 hs ago, was working ok). Slic3r didnt start. So I tried "git pull" . It updated, I tried perl build as usual but the built fails.

lordofhyphens, 2017-03-28:
Well, let's do what the build server is doing then.

Overwrite c:\Strawberry with this:
https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=Strawberry-6.3.0-seg-archive.7z

The reason a recent commit broke things is because the last merge added use
of c++11 features the compiler that was packaged with 5.18 doesn't support
(too old).

Tinchus2009, 2017-03-28:
Joseph, I understand that you are probably really busy with a lot of things, but take a min to read all my report, because if not, at the end you will loose more time. I already said that I tried both ways following your instructions: I tried the MSI file. It didnt worked. Then I tried again overwriting the strawberry installation useing the file you mention. It didnt worked either. Also I downloaded the builted package from your server (it says it was builted like 4 hs ago), well trying to start teh slic3r.exe doesnt work.
If I can gie you some log or any other information I will be more than happy to do so (because right now no version of slicer is wrking on my system)

Tinchus2009, 2017-03-28:
Ok, problem fixed. I have some information it might be usefull to be added to the updated wiki (wich I have just saw and read, and that helped me solve the problem). Steps I followed to install slic3r under win 10 64b:

-installed the slic3r perl
- installing the boos libraries: here it was the problem. All my previous slic3r dev builts were working with boost 1_59_0. I just saw in the wiki that boost 1_63_0 was the version known to work. Well, this is very new since 1_59_0 was working well till last night lol. So I erased the 1_59_0 folder (located under C:\DEV). This folder must be erased, it can not co exist with the neew 1_63_0 installation. After eraseing I followed the boos installation instructions:
-then instal cpanm with: cpan App::cpanminus
-then complete the rest of the wiki steps.

NOTE: built compelted it succesfully, apart from a failed test, but xs installed perfectly. Then I issued the "perl build.pl --gui" command. It went OK. Slic3r started perfectly, I did a couple of slices, no crashes so far. The question is: this build was done from scratch, slic3r folder was brand new, I never aplied the dependencies package, but slic3r gui builted perfectly so far.

lordofhyphens, 2017-03-28:
@Tinchus2009 Ah, right. New compiler means you need to rebuild Boost.
Set BOOST_DIR and the two can coexist.

Tinchus2009, 2017-03-28:
dont be that cruel man! lol
it means that for you expert coders, but not for us simple mortals jajaja

lordofhyphens, 2017-03-28:
It's pretty straightforward on Windows.
For doing it from the command line:
http://www.dowdandassociates.com/blog/content/howto-set-an-environment-variable-in-windows-command-line-and-registry/

From the GUI menus (you will need to close and reopen your cmd.exe or powershell after doing so)
http://www.computerhope.com/issues/ch000549.htm

lordofhyphens, 2017-03-28:
I have no idea why that test is failing.

Tinchus2009, 2017-03-28:
oh man jajajaja. i really preciate that you take your time to give me instructions, and because i preciate that is tgat i repeat: read alllllllllll my pst.... i already solved the problem, aready rebuilted boost. but i think this detail shuold be montion in the wiki.
Thanks a lot!!!

lordofhyphens, 2017-03-28:
@Tinchus2009 It's a wiki, edit away ;-) I don't think as much myself (hence why I just updated this issue).

lordofhyphens, 2017-03-28:
The automatic builds should function again as well at least as of https://github.com/alexrj/Slic3r/commit/d5a0e78c1620d53df79f0c1cbca4ca91108770d0

Tested on Win7 x64.

lordofhyphens, 2017-04-18:
Quietly closing.

closed 2017-04-18 by lordofhyphens