slic3r/Slic3r #3818: Cancel button in the end when using the layer tab is not properly visible

singh1114, 2017-03-28:
### Version
_Version of Slic3r used goes here_
1.3.0-dev

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
798b6c8a08ce812d7a220482308e9d15493a22d3

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 16.04

### Behavior

When we open up the Slicer GUI and switch to the layer's tab, There is a cancel button in the down-right corner which is not properly visible.

[image]

Lenbok, 2017-03-31:
I don't see this problem with current master 8f789da9 on my Ubuntu 16.04, and the GUI has changed quite a bit (for the better!). @singh1114 can you see whether it is fine for you now and close if it's no longer a problem?

xoan, 2017-03-31:
I'm using current master in Arch Linux and the issue is still visible

[image]

This issue is probably here from a long time ago.

lordofhyphens, 2017-03-31:
It's related to your windowing themes. The lettering is barely cut off on my system (GTK3/XFCE).

xoan, 2017-03-31:
Yep, probably related to some wxGTK button padding property.