slic3r/Slic3r #3824: Multiple layers height - proposed automatic transition?

Tinchus2009, 2017-03-30:
This is a feature request:

We have right now the ability into slic3r of haveing different layer height in the same object. I use this feature a lot. It lets you use layer of 0.3 in areas where you dont need any detail, like a wall, and you can configure a layer of 0.1 or less in areas where you want details and smooth areas, like a curved area.
This works great.
But, the transition from a 0.3 layer to a 0.1 layers is done straight forward, there is no transition area. My proposal is to set  an option, so when setting for example layers to 0.3 and then an area of 0.1, I can have X numbers of layers with a transition value, going from 0.3 to 0.1.
What is the advantage of this? Imagine a tower with a semi-sphere in the top: I can print more quicky this tower useing 0.3 layers from the base to where the semi-sphere starts. From there I would use a 0.1 layer to have a smooth rounded area. if the print changes from 0.3 to 0.1, the change will be much noticeable than for example do a transition area of X numbers of layers going from 0.3 to 0.1 layer height.
What do you think?

lordofhyphens, 2017-03-30:
There was discussion by @platsch and @alexrj about replacing or
supplementing the per part layer height setup with the adaptive slicing
feature.

Tinchus2009, 2017-03-30:
adaptive slicing feature? there is such a feature  existing now in slic3r? cant find it....

platsch, 2017-03-30:
I think this is pretty much exactly what we are adressing with adaptive slicing, you can find a lot of detail here: https://github.com/alexrj/Slic3r/pull/1386
it is not merged yet, but the integration continous

lordofhyphens, 2017-03-30:
@platsch has done some great work, I am eager to pull the results when the UI is ready.

Tinchus2009, 2017-03-30:
Yes... it something like this, but what you propose is something more automatic of what I would prefer. And may be it si more compicated of what Im proposeing. Instead of going for algorithms, formulas, etc, it would be more simple to just have more simple control, setting 2 layers heights by the user , then slic3r would just have to compute a transitional area between these two layer heoghts

Tinchus2009, 2017-03-30:
Ok,,, I have read the hole #1386 . I see thaat the work is already advanced. GREAT.
Also I have seen that there is an implementation in the prusa version of slic3r.

closed 2017-03-30 by Tinchus2009

lordofhyphens, 2017-03-30:
There is a limited UI in the current PR, it is being worked on. The way
that it was done in the prusa3d fork is too invasive and doesn't fit the
existing model for how we do dialogs.