slic3r/Slic3r #3827: Slic3r is not deterministic

msrd0, 2017-03-31:
### Version
1.3.0-dev (1.2.9-871-g8f789da9)

### Operating system type + version
ArchLinux

### Behavior
I wanted to print a [3DBenchy](http://www.thingiverse.com/thing:763622) with Slic3r. I was experimenting with the amount of infill when I noticed that something strange happened. This is the version I'd prefer which I get with 20% infill after changing the layer height from 0.2 to 0.15:

[image]

After changing the infill from 20% to 15% I get a completely different result for the solid part:

[image]

When changing back to 20% infill this doesn't get reverted:

[image]

The only way to "restore" the best version is to change layer height to 0.2 and back again. I can reproduce this every time.

By the way to get a Preview at all I need to modify one of the parameters, after loading the STL the Preview and Layers view is empty.

#### STL/Config (.ZIP) where problem occurs
[slic3r-config.zip](https://github.com/alexrj/Slic3r/files/885715/slic3r-config.zip)

lordofhyphens, 2017-03-31:
Is your Perl built with threads?

I generally can't reproduce the layers/preview part of the problem on my Debian system; switching to Preview or Layers is sufficient to start the slicing process (I have background processing turned off).

msrd0, 2017-03-31:
@lordofhyphens I'm using the perl version of my operating system (ArchLinux). The corresponding [PKGBUILD](https://git.archlinux.org/svntogit/packages.git/tree/trunk/PKGBUILD?h=packages/perl#n164) file uses the `-Dusethreads` compile flag. `perl --version` says

    This is perl 5, version 24, subversion 1 (v5.24.1) built for x86_64-linux-thread-multi

Lenbok, 2017-03-31:
I am seeing something like this too. In my case I am setting the exterior brim width, and the number I set in the entry box isn't "sticking" until I change it's value again. I had "Background Processing" unticked in my preferences. Ticking that and restarting makes things seem to work fine (the preview panel gets updates in place with the correct value).

lordofhyphens, 2017-04-02:
More likely something isn't registering as a state change to Slic3r (which would force a state change) and/or one of the methods around Fill() isn't idempotent (which is not the same as not being deterministic).

msrd0, 2017-04-02:
Determinism means that with given prerequisites I can determine the events that will happen in the future. The screenshots I've attached showed that with the same settings (=prerequisites) I get different results, so the event that will happen in the future (after slicing) is different. To me this means that something is not deterministic in that case.

lordofhyphens, 2017-04-02:
In this case, your initial state for the slicing process is not what you
think it is ;-). If it was nondetetermistic you would see different output
from the same settings with a clean state.

It's still a bug that you are seeing that behavior, of course.

alexrj, 2017-04-04:
Thank you for reporting this. I applied a fix, and will work more on it after 1.3.0 is released.

closed 2017-04-04 by alexrj

msrd0, 2017-04-04:
The main problem that slic3r was producing different results with the same settings is gone now. Thanks for fixing this.

After loading an STL file the preview and layers view is still empty. Should I open another issue for this?

lordofhyphens, 2017-04-04:
Always. Separate issues for separate things. Check to make sure it isn't
described by someone else first.

lordofhyphens, 2017-04-04:
@msrd0  #3839