slic3r/Slic3r #3828: Incorrect print on Linux Debian Stretch 4.9.13

androidrobotfactory, 2017-03-31:
### Version
1.2.9 and 1.2.9-780-g4c14c656

### Operating system type + version
Linux host 4.9.0-2-amd64 #1 SMP Debian 4.9.13-1 (2017-02-27) x86_64 GNU/Linux

### Behavior
* Bridge incorrectly added between gear cogs, see images. The one on the top is from Debian Linux Jessie (expected works ok). Note the preview window does not show the problem. Same shape seen GCodeSimulator.jar so does not seem to be a printer problem. Thanks for taking a look.
* Select 
[rip_4.stl.zip](https://github.com/alexrj/Slic3r/files/887124/rip_4.stl.zip)
Sailfish (Makerbot) as G-code flavor, open up rip_4.slt
![img_20170401_095320](https://cloud.githubusercontent.com/assets/4927379/24572290/1a11486a-16c3-11e7-8bad-66824368b566.jpg
)
[image]

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[rip_openscad.zip](https://github.com/alexrj/Slic3r/files/887126/rip_openscad.zip)

lordofhyphens, 2017-04-01:
... what the heck? *sigh*

Stretch is -testing, isn't it? 

Is the preview panel showing anything re:bridge segments in Stretch?

androidrobotfactory, 2017-04-01:
correct Stretch = Debian testing. That screen shot is from Stretch, no bridge. Only shows up when you open the generated gcode in the simulator or on the printer.

xoan, 2017-04-01:
Please, upload your config (`File → Export Config...`).

It seems to be retraction related, probably `Minimun travel after retraction` parameter.

lordofhyphens, 2017-04-18:
[image]

I can't seem to reproduce at all, the image here is from gcode.ws; I just updated this system to Stretch.

The travel moves also don't seem suspect at all. @androidrobotfactory Can you retest with a more recent checkout+build?

alexrj, 2017-05-01:
@androidrobotfactory, the G-code file you attached doesn't show any issue. Moved between teeth are just travel moves (no extrusion). I opened it with multiple preview applications, and even your black screenshot seems to show just travel moves (white).

I think you just need to tune your retraction settings because probably retraction isn't happening. Maybe you want to lower your retraction speed.

I'm closing this issue since the uploaded files do not show any evidence of an issue. If you think we should investigate further, free to reopen including a G-code file which shows the problem.

closed 2017-05-01 by alexrj