slic3r/Slic3r #3829: Small areas may get infill twice, causing over-extrusion

DrLex0, 2017-04-01:
### Version
1.2.9

### Operating system type + version
OS X 10.11.6

### Behavior
* In certain situations, additional solid infill is printed on top of regular infill, causing excessive extruded plastic in those areas to ooze out at the sides if the problem keeps occurring across multiple layers. There seem to be two similar problems, one with internal infill and one with top/bottom layers infill.
* To reproduce: slice the attached model with the attached settings.
* The first problem is invisible in Slic3r's preview. Use a G-code viewer that can show progress within a single layer, and look at the layers between 1.75mm and 3.35mm. The infill for those internal layers is printed twice, once in each direction. The photo below shows the result: so much superfluous material is deposited that it oozes across the outer perimeters.
* The second problem is easily visible in Slic3r: the three topmost layers also have a redundant bit of infill that overlaps with the regular infill.

[image]

[image]

These problems are related to extrusion width. The attached config has width 0 (auto) which causes too wide extrusion, see #3691. Enforcing an infill width of e.g. 0.8mm makes the problems go away for this particular model. However, at the maximum of 1.2mm that #3691 would impose, you'll still find overlapping infill in the model (e.g. in the screw thread part).

The first problem seems to be caused by regular infill still being generated at locations where there is internal support infill already.
The second problem is unclear, maybe it is just because the infill extrusion is wider than the area it is filling…

These problems are still present in the Prusa3d fork of Slic3r.

#### STL/Config (.ZIP) where problem occurs
[ModelAndConfig.zip](https://github.com/alexrj/Slic3r/files/887625/ModelAndConfig.zip)

lordofhyphens, 2017-04-01:
Did you check with a new build of 1.3.0-dev (http://dl.slic3r.org/dev)?

DrLex0, 2017-04-01:
Yes, and the results are the same.

jaggzh, 2017-04-05:
Still happening in a build from yesterday.  I confirmed only the top layer view, in slic3r, has overlapping paths.