slic3r/Slic3r #3835: Background Processing / Live Update not functioning

jaggzh, 2017-04-03:
Fresh build on April 2, 2017: 1.2.9-901-g90623c4
OS: Debian Stable

Description:
Background Processing is enabled, but changes in the Print Settings window do not update until the Print Settings window is closed.

lordofhyphens, 2017-04-03:
I can reproduce this on my Debian stable system and on 1.3.0-dev-90623c4 build for Windows.

alexrj, 2017-04-03:
This happens by design.

lordofhyphens, 2017-04-03:
I believe he is interested in a workflow by which he could move the dialog
over and fiddle with it and watch the print update in the background
(instead of assigning overrides)

jaggzh, 2017-04-04:
Yeah -- it's more convenient to change options and see the results than to close the window to see them, or to select every desired option in the quick settings.

lordofhyphens, 2017-06-03:
One additional change should be made here; when we're not using tabs we should be able to manipulate most of the base UI.

lordofhyphens, 2017-06-03:
fccdd12fba788537673d0faba42c61af54956415 addresses the last part in dialog. Closing.

closed 2017-06-03 by lordofhyphens